wradlib/wradlib #330: Incorrect computation of texture (dp.py) for first range bins?

overeem11, 2019-04-27:
Perhaps I'm overlooking something, but it seems that function texture (dp.py) uses the last range bins for computing the texture for the first range bins. Imagine an array of 360 beams of 1 degree and 802 range bins. If the texture for a first range bin, being closest to the radar, is computed, x2, x5 and x8 seem to contain the values for the last range bin, being furthest from the radar, and hence very far from the first range bin. Hence, it does not make sense to use these in the computation of texture for the 360 first range bins in the example above. It seems that the first column of x2, x5 and x8 in function texture should be made nan.

Best regards,
Aart Overeem, KNMI

kmuehlbauer, 2019-04-28:
@overeem11 Thanks for pointing that out. Seems this is sitting there from right from the beginning. @heistermann This was obviously overlooked at the time of implementation, or do we miss something?

kmuehlbauer, 2019-04-29:
@overeem11 I've a PR ready to fix this. I appreciate your review. Please see #331.

I took the chance to improve performance using numpy views.

kmuehlbauer, 2019-05-10:
Fixed in #331.

closed 2019-05-10 by kmuehlbauer