wradlib/wradlib #335: Install-time dependency: "semver"

s-m-e, 2019-05-17:
From your documentation, it is not entirely clear that `semver` is actually a [install-time dependency](https://github.com/wradlib/wradlib/blob/0b4a35436fb24dfaaaa51519d993888ed212112c/setup.py#L16) of `wradlib`. I.e. `semver` must be installed before you can even attempt to install `wradlib`. As a side-effect, it becomes hard to use `wradlib` as a dependency in yet another package even if I add `semver` as a dependency to my package directly. This is because I can force neither `pip` nor `conda` to install dependencies in a certain order. The only feasible workaround is an ugly hack in my setup scripts to ensure that `semver` is there before I attempt to install my own package with `wradlib` as a dependency. I suggest to make the install work without `semver`.

(This is a variation of #328.)

kmuehlbauer, 2019-05-17:
@s-m-e Yes, it's not completely clear. I'll add this to the docs.

We use `semver` within setup.py to retrieve/parse the wradlib version. If there is a working solution out there without the use of semver, we will gladly consider that.

kmuehlbauer, 2019-05-17:
@s-m-e Nevertheless there are inconsistencies within the whole procedure in setup.py. So thanks for raising that issue. I thought, everything was fine, but it isn't. I'll revise the installation process before the next release. If you have any comments regarding easy drop of `semver` with equivalent workflow, please let me know.

s-m-e, 2019-05-17:
@kmuehlbauer [PEP 518](https://snarky.ca/clarifying-pep-518/) might be an "easy" solution. One of its downsides is that it is not backwards compatible with somewhat older versions of `pip`. A file named `pyproject.toml` can be used to define install-time dependencies (such as `setuptools` for instance).

kmuehlbauer, 2019-05-17:
@s-m-e Yes, I remember reading some of the discussion on PEP 518 back then.

So this means, we create the 'project.toml' file and make sure semver is added there, right?

s-m-e, 2019-05-17:
@kmuehlbauer This is [how I have started doing it](https://github.com/pleiszenburg/gravitation/blob/v0.0.2/pyproject.toml) lately, yes. You would basically add all requirements of your `setup.py` file in there, which in your case is `requires = ["setuptools", "semver"]`. It *appears* to work as expected.

kmuehlbauer, 2019-05-17:
@s-m-e Thanks for clarifying. Seems this is the way to go.

PR #337 (2019-05-24): FIX: install time dependencies
- adapt `pyproject.toml` 
- fix setup.py version handling

closes: #335 

Ping: @s-m-e Can you please review this? At least the part with the `pyproject.toml`? I'll pull this in if you approve, but latest the day after tomorrow.
closed 2019-05-24 by kmuehlbauer