wradlib/wradlib #348: imports

egouden, 2019-06-06:
It is generally recommended not to use "from module import function" to avoid circular imports.
This decreases readability too.

kmuehlbauer, 2019-06-06:
@egouden Thanks for bringing this up!

Seems this is use throughout the project. So what is the suggestion then?

Instead of doing
```
from .projection import proj4_to_osr, reproject
```
we would do
```
from . import projection
```
and prepend `proj4_to_osr` and `reproject` with the module name (`projection`)

kmuehlbauer, 2019-06-06:
Doing a quick search through the code I found these different usages for wradlib-modules:

importing modules
- `import wradlib.ipol as ipol` 
- `from . import trafo as trafo`  (same as `from . import trafo`)

Importing functions from modules
- `from .trafo import decibel`
- `from .io.xarray import create_xarray_dataarray`
- `from . util import import_optional`

For the rewrite of the import sections I would suggest to use relative imports:
- `from . import module`

Functions/Names should not be imported directly, the module should be imported instead.

There are some locations where we already have the import statement within the code to avoid circular dependencies, see https://github.com/wradlib/wradlib/blob/396235eeadf37dc261e37575bd93be2c762dce82/wradlib/util.py#L317 .  

This is open for discussion!

@egouden Can you imagine a more fitting issue title?

egouden, 2019-06-06:
I would suggest using only import "wradlib.package as package" since our project is not that big.

kmuehlbauer, 2019-06-06:
OK! If I got your suggestion right: Since we do not have a deeply nested package is would be more straightforward to go this way:

`import wradlib.clutter as clutter`
`import wradlib.io.dem as io.dem`

egouden, 2019-06-06:
Not sure about how to deal with subpackage...

kmuehlbauer, 2019-09-03:
@egouden Can we apply from this doc: https://docs.python.org/3/tutorial/modules.html#packages ?

egouden, 2019-09-04:
Well this is the general documentation but I cannot find any recommendations regarding our issue.
Could you elaborate?
What I suggested above is what I think is best after a web search on the topic including discussions from the main python developers.
It is also recommended [here](https://docs.python-guide.org/writing/structure/).
I tend to prefer explicit imports of wradlib modules for readability and sustainability of the code but I am OK with relative imports.

PR #381 (2019-09-11): MNT: fix imports, license header, requirements
fixes #348
fixes #375

The imports in wradlib modules follow now the [PEP8 recommendation](https://pep8.org/#imports):
eg.

```python
import os
import sys

import numpy as np
import xarray as xr

from wradlib import io, ipol
from wradlib.georef import projection
```

Absolute imports are used throughout the package.

We would have some name clash with system and wradlib `io` module. This is circumvented using ìmport io as sio`. Other possible name clashes with eg. `gdal` are possible but can also be mitigated.

The function `maximum_intensity_proj [... cut]
closed 2019-09-11 by kmuehlbauer