wradlib/wradlib #349: Speedup zonalstats using geopandas

kmuehlbauer, 2019-06-07:
The zonalstats-module was created having many source points/polygons and less target points/polygons in mind.

For applications where we have much more targets (>>10k), the current approach doesn't scale well. Especially retrieving source indexes per target index and area per target index is terribly slow.

Using geopandas and subsequent grouping and conversion to numpy arrays will speed things up several orders of magnitude.

Unfortunately the GEOS powered GDAL/OGR layer intersection is also quite slow at these high amounts of targets. For intersections with rectangular features there is improvement ahead (see https://github.com/OSGeo/gdal/issues/1209), although not implemented yet.

Suggestion: Use `geopandas` as optional dependency and change the zonalstats-code accordingly.

egouden, 2019-06-07:
The zonalstats module has always been intriguing to me because it is so general ;-)
I was frustrated not finding these features in GDAL/OGR.
I found rasterstats but it was somewhat too high-level.
Could you elaborate which application you are thinking and why geopandas will be faster?

Recently I looked at function to convert raster values to sweep polygons (for beamblockage).
zonalstats was indeed too slow and I end up using bin statistics with pixel centers.
I think the latter method could be easily extended to pixel edges.
This is almost ready for a PR.

kmuehlbauer, 2019-06-07:
The thing is, that the creation of source and target points/polygons is reasonably fast. Taking into account, that these  two layers normally won't change (radar bins, river catchments, model domain etc) this is acceptable.

The GEOS/GDAL powered layer intersection degrades (no multithreading) whith increasing target feature count. This also is not a neckbreaker considering a one time computation.

What I was missing in the current implementation is the reloading of the indices and weights from the created shape file. The GDAL/Ogr based feature iteration is slow taking approx 20ms per feature.

I explored geopandas for the extraction of indices and weights and found that this is faster by a factor of 100.

The application is weighted areal rainfall summation from groundradar bins  over a model grid. Ground radar 360x600 (250m range res) and RADOLAN grid 1km grid spacing.

kmuehlbauer, 2019-06-10:
While reading GDAL/OGR documentation I found this [here](https://gdal.org/drivers/vector/memory.html) for the Memory driver:

*The driver does not implement spatial or attribute indexing, so spatial and attribute queries are still evaluated against all features. Fetching features by feature id should be very fast (just an array lookup and feature copy).*

So this might be something which was overlooked at the time of implementation. Since we copy the source layer to a memory layer each time we reload the vector shape file, this might have degrading effect on the attribute filter queries.

closed 2019-06-11 by kmuehlbauer

kmuehlbauer, 2022-03-17:
Just for reference, geopandas accessor is implemented for VectorData in #558

PR #351 (2019-06-11): ENH: Speedup ZonalStatistics
This PR consist of three different enhancements to speed up processing:

- use temporary files using '/vsimem` to be able to use indexing
- create spatial and attribute index files prior processing
- retrieve attrributes and geometry properties within one function

closes #349