wradlib/wradlib #353: 'WRADLIB_DATA' path '' does not exist

rvalenzuelar, 2019-06-24:
I'm using the partial beam blocking example with a custom DEM file. I'm not sure why I'm getting this error if I'm not using the files from the example. 

I thought exporting an empty entry for WRADLIB_DATA would help but I get the same message.

kmuehlbauer, 2019-06-24:
@rvalenzuelar Can you show the code? Normally you can just use the filename of your custom dem without setting WRADLIB_DATA (you must not use get_wradlib_data_file then).

rvalenzuelar, 2019-06-24:
Ok, I know what happened. I have another mac with wradlib and first I used the example and then changed to custom DEM (with WRADLIB_DATA already exported). In this mac I'm using the same python script and started with the custom DEM BUT kept using the get_wradlib_data_file function. 

What function should I use instead?

kmuehlbauer, 2019-06-24:
You just feed the dem filename (full path) to the open_raster function.

rvalenzuelar, 2019-06-24:
Perfect! Thanks!

closed 2019-06-24 by rvalenzuelar