wradlib/wradlib #373: Interpolation kwarg not passed to interpolator call method

jorahu, 2019-08-03:
Using Nearest interpolator with maxdist kwarg when using wradlib.comp.togrid function does not work although togrid documentation says kwargs are passed to interpolator class. 
https://github.com/wradlib/wradlib/blob/master/wradlib/comp.py  Line 90 data_on_subgrid = ip(data).reshape((len(ix))) calls the interpolator without maxdist kwarg. 
When calling the comp.togrid method with maxdist gives TypeError: TypeError: __init__() got an unexpected keyword argument 'maxdist'.

kmuehlbauer, 2020-03-12:
@jorahu Sorry, this had to be postponed to the next (bugfix-) release. It's a bit more complicated thatn anticipated and might need an rewrite of the interpolator classes.

kmuehlbauer, 2020-05-26:
@jorahu In #425 I've implemented a bit of a hack to get this in. You can add the calling keyword arguments by adding
```python
togrid(...,...,..., call_kwargs=dict(maxdist=10))
```
This will be available in wradlib 1.7 in about two weeks time.

PR #425 (2020-05-26): FIX: introduce 'call_kwargs' in comp.togrid 
fixes #373
closed 2020-05-26 by kmuehlbauer