wradlib/wradlib #382: Possible error on the notebook/documentation

ElmerJeanpierreLopez, 2019-09-15:
In the notebook of BeamBlockage mentions that angle of elevation should be in deg. But, the documentation of the function sweep_centroids (https://docs.wradlib.org/en/latest/generated/wradlib.georef.polar.sweep_centroids.html) that use after says that elevation angle must be on rad. So, one of them there must be wrong.

kmuehlbauer, 2019-09-15:
@ElmerJeanpierreLopez Thanks for reporting this issue. This is an error in the documentation of `sweep_centroids`. It should read `degrees`.

Would you be able to issue a PR to fix this?

ElmerJeanpierreLopez, 2019-09-15:
Could be. But I don't know what means PR.

ElmerJeanpierreLopez, 2019-09-15:
I feel so embarrassing. I already created a Pull Request of this.

closed 2019-09-15 by ElmerJeanpierreLopez

kmuehlbauer, 2019-09-16:
> I feel so embarrassing. I already created a Pull Request of this.

No worries, thanks for fixing this, your help is very much appreciated!