wradlib/wradlib #386: georef init

egouden, 2019-09-16:
Why not using from module import * in the georef init?
The current situation is really time consuming when developing.

kmuehlbauer, 2019-09-16:
I find it easier to maintain a stable API (in terms of functions). Another point if we would do this (eg `from .polar import *`) all other packages imported within the subpackage will clutter the namespace (like numpy, warnings etc).

egouden, 2019-09-16:
OK.
What about keeping only the most useful functions?

kmuehlbauer, 2019-09-16:
If we identify functions which should not be used via `wradlib.georef.function` (but can still be reached via `wradlib.georef.subpackage.function`) we might remove them from `georef`. 

Which do you have in mind?

egouden, 2019-09-17:
I think my last suggestion makes things complicated.
In fact as already mentioned I prefer not to use the georef submodules.
Therefore I am fine with the current situation.
I just hope there is a way to automatically import only the functions.

kmuehlbauer, 2019-09-17:
@egouden Yes, I'm still looking for a solution to this. If you fiond anything by chance, let me know.

kmuehlbauer, 2019-09-17:
@egouden I have an idea for a solution:

In the submodule we do something like this:

```python
"""
Miscellaneous
^^^^^^^^^^^^^

.. autosummary::
   :nosignatures:
   :toctree: generated/

   {}
"""
__all__ = ['bin_altitude', 'bin_distance', 'site_distance']
__doc__ = __doc__.format('\n   '.join(__all__))
```

We have to specify the functions which should be exported/documented in the `__all__` variable of the respective submodule. This will then also be added to the docstring.

In the module we would then be able to use `from .misc import *` etc..

I tested this and can issue a PR if you think this will be more easy to use.

kmuehlbauer, 2019-09-20:
resolved by #388

closed 2019-09-20 by kmuehlbauer