wradlib/wradlib #389: xarray: computation over bin timeseries

egouden, 2019-09-17:
I am struggling for hours to apply a function on each radar bin timeseries.
I noticed that the xarray interface returns a 2d array with (time, range) when providing several volume files.
From that I can't find a way to do what I want.
apply_ufunc seems a good candidate but it is very hard to understand.
I think it would be easier to start form a 3d xarray with (volume, azimuth, range).

kmuehlbauer, 2019-09-17:
@egouden Use `dim0='azimuth'` when calling.

egouden, 2019-09-17:
I get a MergeError...

kmuehlbauer, 2019-09-17:
The reader itself can't handle multiple timestep from the same elevation. It is designed to handle single volumes as laid out in Odim/Cfradial standards. If you want to generate timeseries you would need to read multiple volumes and merge the wanted xarray objects afterwards. I already wrapped my head around this for a proper solution, but it is hard to solve. We could implement some TimeSeries class which takes several volumes.

egouden, 2019-09-17:
I think this would be very useful indeed.
The current implementation already provides some way of using multiple timesteps volumes.
I will dig a bit further.

egouden, 2019-09-24:
Well it works with homogeneous volumes when the time dimension is used.
With the azimuth dimension, you just need to use concat instead of merge.
For inhomogenous volumes it could work but one would have to check the root for each sweep.
I don't understand why you would have the same sweep in the current use case?

kmuehlbauer, 2019-09-24:
@egouden A simple code example showing your current workflow would be nice. I have some problems to follow up...

kmuehlbauer, 2019-10-29:
should be resolved by #395

closed 2020-03-12 by kmuehlbauer