wradlib/wradlib #400: Coordinates for KNMI Radar composite

meteoDaniel, 2020-01-06:
Dear all, 

I am using wradlib to read KNMI h5 composite reflectivity data. Now I need the coordinates to the pixel values. 

I tried to use some wradlib functions but I was not successfull. 
The h5 file provides the following information: 
```
{'projection_indication': b'Y',
 'projection_name': b'STEREOGRAPHIC',
 'projection_proj4_params': b'+proj=stere +lat_0=90 +lon_0=0 +lat_ts=60 +a=6378.14 +b=6356.75 +x_0=0 y_0=0'}
{'geo_par_pixel': b'X,Y',
 'geo_dim_pixel': b'KM,KM',
 'geo_pixel_def': b'LU',
 'geo_number_columns': array([700], dtype=int32),
 'geo_number_rows': array([765], dtype=int32),
 'geo_pixel_size_x': array([1.0000035], dtype=float32),
 'geo_pixel_size_y': array([-1.0000048], dtype=float32),
 'geo_column_offset': array([0.], dtype=float32),
 'geo_row_offset': array([3649.982], dtype=float32),
 'geo_product_corners': array([ 0.      , 49.362064,  0.      , 55.973602, 10.856453, 55.388973,
         9.0093  , 48.8953  ], dtype=float32)}
```

So I hope anyone can tell me what I have to do. Thanks a lot. 

Data can be accessed via:[KNMI Data]( https://data.knmi.nl/datasets/radar_reflectivity_composites/2.0)

kmuehlbauer, 2020-01-07:
@meteoDaniel 

You can derive the projection using this snippet:
```python
lat_ts = 60. # taken from lat_ts value of given projstr
k0 = (1. + np.sin(np.radians(lat_ts))) / 2.
# adapted projstr (taken from projection_proj4_params)
projstr = f'+proj=stere +lat_0=90 +lon_0=0 +k_0={k0:8.10f} +a=6378140 +b=6356750 +x_0=0 y_0=0 +units=km'
print(projstr)
proj_stereo = wrl.georef.proj4_to_osr(projstr)
```

Now you can transform into any other projection (like latlong):

```python
projstr = f'+proj=latlonh +a=6378140 +b=6356750 +x_0=0 y_0=0'
proj_latlong =  wrl.georef.proj4_to_osr(projstr)

```

kmuehlbauer, 2020-01-07:
One thing which is not really clear to me, the given corner points are close to the calculated values but not equal. Not sure, where this comes from. Maybe the values KNMI has in there are not quite fitting, ping @leijnse.

meteoDaniel, 2020-01-07:
Dear @kmuehlbauer , thanks for sharing this snippet. I tried this with the raw proj4 string and with your adapted string but in both ways it raises : 

```
AttributeError: 'SpatialReference' object has no attribute 'Fixup'
```

I am working with wradlib==1.2.0 in a docker container

kmuehlbauer, 2020-01-07:
@meteoDaniel You would need wradlib 1.5 I think and/or a newer gdal (>=3.0). I can try to get this running for 1.2. What gdal and proj versions have you running?

meteoDaniel, 2020-01-07:
@kmuehlbauer I am testing version 1.5 actually. I am using osgeo:gdal as base image. So I should run on the latest gdal version. 

If I am using your string, I am getting the following error:
```
+proj=stere +lat_0=90 +lon_0=0 +k_0=0.9330127019 +a=6378140 +b=6356750 +x_0=0 y_0=0 +units=km
Traceback (most recent call last):
  File "bin/create_radar_images.py", line 5, in <module>
    from src.radarImages import CreateRadarImages
  File "/app/src/radarImages.py", line 15, in <module>
    from src.radarDataHandler import RadarDataHandler
  File "/app/src/radarDataHandler.py", line 196, in <module>
    proj_latlong =  wrl.georef.proj4_to_osr(projstr)
  File "/usr/local/lib/python3.6/dist-packages/wradlib/georef/projection.py", line 180, in proj4_to_osr
    raise ValueError("proj4str validates to 'ogr.OGRERR_CORRUPT_DATA'"
ValueError: proj4str validates to 'ogr.OGRERR_CORRUPT_DATA'and can't be imported as OSR object

```
But It works fine with another proj4.string. 
Thanks a lot

closed 2020-01-07 by meteoDaniel

kmuehlbauer, 2020-01-07:
@meteoDaniel Glad it works now. Could you share the working projstr?

meteoDaniel, 2020-01-07:
I took that one to reporject: 

```
projstr = f'+proj=longlat +datum=WGS84 +no_defs '
```