wradlib/wradlib #404: Importing wradlib leads to hdf5 errors when using xarray

GroovyGregor, 2020-02-05:
Importing wradlib leads to hdf5 errors when using xarray. 

To reproduce the error, I tested the installation on fresh docker instances based on the current `ubuntu` and `ubuntu:latest`. 

```
docker run -it --rm ubuntu bash
```

The following commands:
```
apt update && apt install -fy python3-pip python3-gdal 
pip3 install semver requests scipy 
pip3 install wradlib
cat > test.py <<EOF
import wradlib
import xarray as xr

D = xr.Dataset({'test': (['x'], [1,2,3])}) 
D.to_netcdf('dummy.nc')
EOF
python3 test.py
```

lead to the following errors:
```
root@b42885602b6e:/# python3 test.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/api.py", line 1089, in to_netcdf
    dataset, store, writer, encoding=encoding, unlimited_dims=unlimited_dims
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/api.py", line 1135, in dump_to_store
    store.store(variables, attrs, check_encoding, writer, unlimited_dims=unlimited_dims)
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/common.py", line 298, in store
    variables, check_encoding_set, writer, unlimited_dims=unlimited_dims
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/common.py", line 339, in set_variables
    writer.add(source, target)
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/common.py", line 188, in add
    target[...] = source
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/netCDF4_.py", line 51, in __setitem__
    data[key] = value
  File "netCDF4/_netCDF4.pyx", line 4950, in netCDF4._netCDF4.Variable.__setitem__
  File "netCDF4/_netCDF4.pyx", line 5229, in netCDF4._netCDF4.Variable._put
  File "netCDF4/_netCDF4.pyx", line 1887, in netCDF4._netCDF4._ensure_nc_success
RuntimeError: NetCDF: HDF error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 5, in <module>
    D.to_netcdf('dummy.nc')
  File "/usr/local/lib/python3.6/dist-packages/xarray/core/dataset.py", line 1545, in to_netcdf
    invalid_netcdf=invalid_netcdf,
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/api.py", line 1104, in to_netcdf
    store.close()
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/netCDF4_.py", line 492, in close
    self._manager.close(**kwargs)
  File "/usr/local/lib/python3.6/dist-packages/xarray/backends/file_manager.py", line 221, in close
    file.close()
  File "netCDF4/_netCDF4.pyx", line 2485, in netCDF4._netCDF4.Dataset.close
  File "netCDF4/_netCDF4.pyx", line 2449, in netCDF4._netCDF4.Dataset._close
  File "netCDF4/_netCDF4.pyx", line 1887, in netCDF4._netCDF4._ensure_nc_success
RuntimeError: NetCDF: HDF error
```

Uncommenting the wradlib import solves the issue.

Do you have any suggestions or fixes? 
Many thanks,
Gregor

GroovyGregor, 2020-02-05:
I just tried with the current master version from github:

    pip3 install git+https://github.com/wradlib/wradlib.git
    Successfully installed wradlib-1.6.0.dev20

Still the same issue

kmuehlbauer, 2020-02-05:
@GroovyGregor Can you give me the full output of the pip installation log? I'm assuming some weird hdf5/netcdf issue or some problem with pip dependencies resolving. This is still a problem installing gdal with all its dependencies into system python. Also can you check the version of h5py (including library) and netcdf4 (including library) and xarray ( `xr.show_versions()`). Thanks.

kmuehlbauer, 2020-02-06:
@GroovyGregor Same is happening if you issue this:

```bash
apt update && apt install -fy python3-pip python3-gdal 
pip3 install semver requests scipy 
pip3 install wradlib
cat > test.py <<EOF
import h5py
import xarray as xr

D = xr.Dataset({'test': (['x'], [1,2,3])}) 
D.to_netcdf('dummy.nc')
EOF
python3 test.py
```
This seems like some weird h5py/hdf5/netcdf4 dependency hole and is not specific to wradlib.

kmuehlbauer, 2020-02-06:
But this works:

```bash
```bash
apt update && apt install -fy python3-pip python3-gdal 
pip3 install semver requests scipy 
pip3 install wradlib
cat > test.py <<EOF
import netCDF4
import h5py
import xarray as xr

D = xr.Dataset({'test': (['x'], [1,2,3])}) 
D.to_netcdf('dummy.nc')
EOF
python3 test.py
```
```

kmuehlbauer, 2020-02-06:
Closing, since this is not wradlib related but most likely some incompatibility with netcdf/hdf5. A workaround with importing netCDF4 before anything else is proposed, see above.

closed 2020-02-06 by kmuehlbauer

meteoDaniel, 2020-06-04:
Dear all, find further information here: 
https://github.com/Unidata/netcdf4-python/issues/653

meteoDaniel, 2020-06-04:
@kmuehlbauer , I recommend to use h5netcdf instead of h5py engine because this avoids problems with incompatibility between h5py and netcdf4 and the performance is much better.

kmuehlbauer, 2020-06-04:
@meteoDaniel I'm not sure how h5netcdf will help here, it's based on h5py too.

kmuehlbauer, 2020-06-04:
@meteoDaniel Ah, you mean as a replacement for netcdf4?