wradlib/wradlib #411: Google colab: env variable

kunalrustagi08, 2020-03-10:
Hi. I am trying to use wradlib on google colab, but unable to set the environment variable for WRABLIB_DATA. 

I tried the way it was described in [this link](https://docs.wradlib.org/en/stable/jupyter.html#how-can-i-get-the-example-data), but got an OSError.
[image]
[image]

Can someone please help me how to set it using %env.

kmuehlbauer, 2020-03-10:
@kunalrustagi08 You would need to replace the 'full/path/to/blabla' with the actual path where wradlib-data is installed.

kunalrustagi08, 2020-03-11:
> @kunalrustagi08 You would need to replace the 'full/path/to/blabla' with the actual path where wradlib-data is installed.

Sorry, but I am not able to find the path of wradlib_data in google colab. Can you please help?

kmuehlbauer, 2020-03-11:
It's not there in colab until you install/upload it.

kunalrustagi08, 2020-03-11:
I have successfully installed wradlib using pip, still getting the error.
[image]

kmuehlbauer, 2020-03-11:
@kunalrustagi08 You need to follow the docs. In the link you provided in your first post, there is written:

*Most notebooks use example data. These data are provided in a separate repository and you need to “install” them manually. “Installing” actually means: Download the data archive [here](https://codeload.github.com/wradlib/wradlib-data/zip/master) and extract it into any arbitrary directory. Now you need to set an environment variable pointing to that directory:*

You need to download the data archive supplied within the link, extract it to some location where you can access it and then point `WRADLIB_DATA` env variable to this location.

kunalrustagi08, 2020-03-11:
Hey. I just realized my mistake that I overlooked the simple instructions. Thank you for pointing it out to me. Now it is working fine. :)

kmuehlbauer, 2020-03-11:
@kunalrustagi08 Glad you got it working. Next time, if you have some user question which is not directly related to the code itself, please use the [wradlib-user group](https://groups.google.com/forum/#!forum/wradlib-users).

closed 2020-03-11 by kmuehlbauer

kunalrustagi08, 2020-03-11:
> @kunalrustagi08 Glad you got it working. Next time, if you have some user question which is not directly related to the code itself, please use the [wradlib-user group](https://groups.google.com/forum/#!forum/wradlib-users).

Yes. Definitely!