wradlib/wradlib #417: Error in reading a shapefile in a unix system

IreneCrisologo, 2020-03-17:
Hi,

I am getting this error when trying to read a shapefile in a Unix server:

`KeyError: 'Illegal field requested in GetField()'`

However I can read the same file in a Windows machine. 

The snippet im using to read it is:

**dataset, inLayer = wrl.io.open_vector(fname_shp)
borders, keys = wrl.georef.get_vector_coordinates(inLayer, key='OBJECTID')**

I get the error even on a freshly made environment. :( 

Any leads would be appreciated!

kmuehlbauer, 2020-03-18:
@IreneCrisologo First, can you check the versions of GDAL on both systems? And second, can you provide the full ErrorMessage? Third, are you able to provide such a shapefile for testing?

I'm thinking that this issue depends on the GDAL version.

IreneCrisologo, 2020-03-18:
Hi Kai,

I've tried it now on 4 different environments--two on my laptop (windows) and two on the Unix server.
The GDAL versions on my laptop are 2.4.1 for the old environment and 3.0.4. for the freshly installed one. The script works fine in both environments.

On the server, in the old environment it's 2.4.2 and in the new environment it's 3.0.4. The script doesn't work on both environments.

The full error message (on the new environment in the server):

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-7-5459e2b7e307> in <module>
      1 dataset, inLayer = wrl.io.open_vector(fname_shp)
----> 2 borders, keys = wrl.georef.get_vector_coordinates(inLayer, key='objectid')

~/.conda/envs/radar/lib/python3.7/site-packages/wradlib/georef/vector.py in get_vector_coordinates(layer, **kwargs)
    148         if feature:
    149             if key:
--> 150                 attrs.append(feature[key])
    151             geom = feature.GetGeometryRef()
    152             if dest_srs:

~/.conda/envs/radar/lib/python3.7/site-packages/osgeo/ogr.py in __getitem__(self, key)
   4694                 fld_index = self.GetGeomFieldIndex(key)
   4695             if fld_index < 0:
-> 4696                 raise KeyError("Illegal field requested in GetField()")
   4697             else:
   4698                 return self.GetGeomFieldRef(fld_index)

KeyError: 'Illegal field requested in GetField()'
```

I investigated some more, and found that if I don't provide the `key` parameter in the function, it is able to read it, but it leaves the `keys` array in the output empty. However, that `keys` array is a useful output.

The shapefile in question is available here: https://drive.google.com/open?id=1yYvDuAyTjMWxt86Je_EtS74UV2IhctYU

Thanks,
irene

kmuehlbauer, 2020-03-18:
@IreneCrisologo Thanks for the feedback. So it's no GDAL version problem, but some linux/unix issue.

I'll have a look tomorrow...

kmuehlbauer, 2020-03-18:
@IreneCrisologo I have the same issues on my linux box and on my windows laptop (new env).

Are you sure, that you have the same files on your systems which you provided in your link. Shapefiles sometimes have more ancillary files (like the .shx index file).

IreneCrisologo, 2020-03-18:
This is embarrassing but yes that was indeed the problem, the shapefile in the server was missing the ancillary files. Thank you for testing this out!

closed 2020-03-18 by IreneCrisologo

kmuehlbauer, 2020-03-18:
@IreneCrisologo Been there, done that. :eyes: Happens all the time. Glad it works now. :grin: