wradlib/wradlib #426: Spherical_to_proj - error

figelion, 2020-05-29:
Hello
I started to have a problem with georef.spherical_to_proj. After some changes, which were just adding a few radar files in xarray i have parameters 
```
r  = np.array(range(1,942))
az = np.array(range(0,360))
th = np.array([2]*359)
'radar_location = [0., 0., 0.]'
```
After putting them into the function:
```
import wradlib as wrl
geog_coords_polar = wrl.georef.spherical_to_proj(
            r,
            az,
            th,
            sitecoords=radar_location,
        )
```
Putting this in jupyter i get this error:
> TypeError: in method 'CoordinateTransformation_TransformPoints', argument 1 of type 'OSRCoordinateTransformationShadow *'

wrl.georef.spherical_to_xyz is working without problems.

but while I was using this in script with the same parameters  I got stuck at this function. Earlier I wasn't using xarray. I was getting parameters from the raw file (parameters had ndarray type). The only difference between both is az. The values were not ordered (in the raw file) from 0 to 360, and were floats.

kmuehlbauer, 2020-05-29:
@figelion 

Are you using wradlib from conda/conda-forge by chance? If not, how did you install wradlib? I suspect some problem with GDAL+PROJ.

figelion, 2020-05-29:
From conda-forge. I will create new env and see how it will go. Thanks for fast replay.

kmuehlbauer, 2020-05-29:
@figelion If anything goes wrong, also install `proj-data`. I just today had the problem myself, that some of the projection information was missing since conda-forge moved this out to `proj-data`.

kmuehlbauer, 2020-06-02:
@figelion  Did you, and if, how did you solve the issue?

figelion, 2020-06-02:
@kmuehlbauer Yes, I solved it. It turned out that it was my bad. I even made the same error in the description above. After concate and adding together a few xarrays the elevation was lost so I got it from the first file.  And the effect was that elevation had 359 elements and azimuth 360. After adding 1 element to the elevation it worked. I was misled a little because spherical_to_xyz computed, so I thought that maybe there is some problem with proj function.

closed 2020-06-02 by figelion