wradlib/wradlib #433: Docker Images throw error on RADLON data

kenodressel, 2020-06-22:
Using the example notebooks provided in the docker Image I get the following error:
`AttributeError: 'NoneType' object has no attribute 'TransformPoints'` here:
https://github.com/wradlib/wradlib/blob/d329019a487608ec91559e9606481f10dd56d3d4/wradlib/georef/projection.py#L317

#### MCVE Code Sample
Basically follow the normal docker setup, execute the notebook command and include the notebook & data folders as volumes. Then run any code containing:

```python
wradlib.georef.rect.get_radolan_grid
```

#### Expected Output
 This should not error.

#### Problem Description

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>
1.5.0
</details>
This is, AFAIK, an outdated version and might be fixed in newer setups but since it is the one contained in the official docker images it is still a relevant problem.

kmuehlbauer, 2020-06-22:
@kenodressel Thanks for reporting this. The docker images are outdated unfortunately. This will take some time to fix them. I'll report back when this is done!

kmuehlbauer, 2020-06-24:
@kenodressel Could you please check the images in [this docker-repo](https://hub.docker.com/r/wradlib/docker-test/tags)?

- 1.7.1-min: wradlib environment)
- 1.7.1-full: wradlib plus notebook plus devel dependencies

The workflow will be more automated and tests of the images have to be added, but I would appreciate any comments on the images so far. The error you described was due to the fact, that the conda environment was not correctly activated on container creation. This should be OK now with the test images. All other images have been removed.

kenodressel, 2020-06-25:
I checked the image and the version looks good but I am still getting the same error. So it might be an application bug rather than a docker bug.
using the notebook: `/notebooks/radolan/radolan_showcase.ipynb` and executing the second cell with code:

```
# Get coordinates
radolan_grid_xy = wrl.georef.get_radolan_grid(900,900)
radolan_egrid_xy = wrl.georef.get_radolan_grid(1500,1400)
radolan_wgrid_xy = wrl.georef.get_radolan_grid(1100, 900)
(.....)
```
I get 
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-2-9ddaf9dce980> in <module>
      1 # Get coordinates
----> 2 radolan_grid_xy = wrl.georef.get_radolan_grid(900,900)
      3 radolan_egrid_xy = wrl.georef.get_radolan_grid(1500,1400)
      4 radolan_wgrid_xy = wrl.georef.get_radolan_grid(1100, 900)
      5 x = radolan_grid_xy[:,:,0]

/opt/conda/envs/wradlib/lib/python3.8/site-packages/wradlib/georef/rect.py in get_radolan_grid(nrows, ncols, trig, wgs84)
    207     res = griddefs[(nrows, ncols)]["res"]
    208 
--> 209     x_0, y_0 = get_radolan_coords(9.0, 51.0, trig=trig)
    210 
    211     x_arr = np.arange(x_0 - j_0, x_0 - j_0 + ncols * res, res)

/opt/conda/envs/wradlib/lib/python3.8/site-packages/wradlib/georef/rect.py in get_radolan_coords(lon, lat, trig)
     66         proj_wgs = projection.get_default_projection()
     67 
---> 68         x, y = projection.reproject(
     69             lon, lat, projection_source=proj_wgs, projection_target=proj_stereo
     70         )

/opt/conda/envs/wradlib/lib/python3.8/site-packages/wradlib/georef/projection.py in reproject(*args, **kwargs)
    315     else:
    316         ct = osr.CoordinateTransformation(projection_source, projection_target)
--> 317     trans = np.array(ct.TransformPoints(C))
    318 
    319     if len(args) == 1:

AttributeError: 'NoneType' object has no attribute 'TransformPoints'
```

It should be noted that the above error was produced on windows while the one below is on Arch Linux. While this should usually not make a difference in docker, sometimes it can :)

kmuehlbauer, 2020-06-25:
@kenodressel Just to make this bulletproof. Could you start the container into bash and start the notebook server from the shell? Does the problem persist. It seems the activation problem is still not solved.

kmuehlbauer, 2020-06-25:
@kenodressel 
Starting the notebook server from the container terminal works for me with the above notebook.
It seems, the conda-activation scripts are not called and the `PROJ_LIB` and `GDAL_DATA` environment variables are not set. 

We need to figure out, how to correctly activate the conda-environment using su-exec.

kmuehlbauer, 2020-06-25:
@kenodressel So, the container init runs now through a dedicated start script where the environment is activated. The new test images will be uploaded by CI shortly. I very much appreciate if you could test again.

kmuehlbauer, 2020-06-25:
@kenodressel The latest test images are working now as expected. But the documentation will need an update. Please close this one, if it works for you too.

kenodressel, 2020-06-25:
The new images work as expected for me too. I only tested the one image but it seems to be working as expected.

kmuehlbauer, 2020-06-29:
@kenodressel I'm closing this one. 

The images `master-min` and `master-full` are build daily via TRAVIS-CI cronjob, so they will have the latest available conda-forge packages and the latest wradlib github-master. The images `X.Y.Z-min` and `X.Y.Z.-full` are build by TRAVIS too and will contain the available conda-forge packages at build time and the wradlib X.Y.Z tagged version.

The docs will be adapted with the next wradlib version.

closed 2020-06-29 by kmuehlbauer