wradlib/wradlib #439: Utility for Cross section plots

pandasambit15, 2020-07-08:
Hi!

Is it possible to have an utility for plotting cross sections of radar fields similar to what is offered by PyART. I am avoiding PyART because it doesn't support all file formats (as compared to wradlib). Moreover, wradlib is much more open and versatile to use. So, would really appreciate if it can be implemented.

Example from PyART gallery: http://arm-doe.github.io/pyart/source/auto_examples/plotting/plot_xsect.html

Regards,
Sambit

#### MCVE Code Sample

```python
# Your code here

```

#### Expected Output

#### Problem Description

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

</details>

kmuehlbauer, 2020-07-08:
@pandasambit15 For this we would need radar volumes to create kind of fake RHI. What format is your radar data?

pandasambit15, 2020-07-08:
Hi!

We have PPI volume scan data and 3D CAPPIs created using the vpr utility.

Actually, you don't need RHI exactly, PPI cross sections and CAPPI cross
sections can be utilized I think.

Another way of implementation is like wrf-python, where they use model
fields of radar reflectivity:
https://wrf-python.readthedocs.io/en/latest/plot.html#cross-example

Regards,
Sambit

kmuehlbauer, 2020-07-08:
> Actually, you don't need RHI exactly, PPI cross sections and CAPPI cross sections can be utilized I think.

That's why I referred to "fake" RHI. :grin: Do you need also arbitrary cuts which do not pass over the radar?

pandasambit15, 2020-07-08:
> That's why I referred to "fake" RHI. 😁 Do you need also arbitrary cuts

Yes actually! That would be great!

For example, in case of cyclones, if one wants to take a transect across
the eye wall region, or different regions of interest which are not
necessarily passing over the radar site. This would be really helpful!

I can share some papers if you'd like some reference!

Regards,
Sambit

kmuehlbauer, 2020-07-09:
@pandasambit15 Let's talk a bit about API. 

From my perspective we have two different data sources, polar volumes and cartesian volumes.

What should the user give as parameters for the two data sources?

- Polar Volumes
    - 3D polar np array, corresponding list of elevations, radar site coordinates
    - (or) list of 2D polar np arrays, corresponding list of elevations, radar site coordinates
    - coordinates of slice, either (az, range) or (lon, lat)
- Cartesian Volume
    - 3D cartesian np array,  corresponding coordinates
    - coordinates of slice, either (x, y) or (lon, lat)

Is there something missing? What else need to be considered?

pandasambit15, 2020-07-09:
Hi Kai!

I think these parameters look sufficient for now, considering the user
wants to view a cross section of one measurement record (one scan).

Additionally, if one needs to check the time evolution of the cross section
or transect, say from a time series ncfile with [x,y,z,t] or [lon, lat,
alt, t], then may be an option to include (time, height) range would be
helpful. Although, the user has to provide the data in a time series format
of the dimensions mentioned above.

Something like these:
https://www.researchgate.net/figure/Mean-reflectivity-time-height-a-and-time-series-b-plot-of-CAPPI-Z-at-27-th-April-at_fig4_334895810

https://www.researchgate.net/figure/Time-height-plot-of-the-equivalent-radar-reflectivity-factor-Z-e-measured-by-MRR_fig2_303883686

But yeah, these are additional, very specific requirements (mainly for
publication purpose).

Regards,
Sambit

*Sambit Kumar Panda*
*Scientist/Engineer-SC, ASD/AOSG/EPSA,*
*SAC(ISRO), Ahmedabad*

TinoManzato, 2022-02-14:
Sorry for asking after so much time. 
At the end, were you able to plot a vertical cross section (fake RHI) from a PPI volume using wradlib?

kmuehlbauer, 2022-02-14:
@TinoManzato Unfortunately not, as nobody implemented it yet. I think it's more or less straightforward. The major pain point is how to handle the spaces between the sweeps which are not measured.

JulianGiles, 2023-01-05:
We have added a new function `wradlib.util.cross_section_ppi` to implement this feature. #610 
The function has the option to generate artificial empty rays to plot the spaces between sweeps, and also the option to extract data along an arbitrary line defined by two points.
Thanks @kmuehlbauer for providing a base code for me to build on!

kmuehlbauer, 2023-02-13:
available in 1.19

closed 2023-02-13 by kmuehlbauer