wradlib/wradlib #441: Plotting Scan Strategy

kmuehlbauer, 2020-07-17:
This is a follow up on the conversation at [wradlib-users group](https://groups.google.com/g/wradlib-users/c/Vud23QpQtmo/m/ni-e_biVBAAJ).

The plotting of the scan strategy is quite ancient and needs update. While implementing the version provided by @jorahu into wradlib code-base some design decisions have to be made.

From my perspective there should be two different modes:

- plotting on normal cartesian grid
- plotting on curvelinear grid

Either of them should have a switch to render the orographie into the plot.

Now, the calling signature of `plot_scan_strategy` needs adaption.

Status Quo:

```python
def plot_scan_strategy(ranges, elevs, site, vert_res=500.0, maxalt=10000.0, ax=None):
``` 
Proposed:

```python
def plot_scan_strategy(ranges, elevs, site, vert_res=500., maxalt=10000., ax=None,
                       az=None, beamwidth=1.0, range_res=None, maxrange=None, units='m',
                       terrain=None, cg=False, ):
```

- az - azimuth angle to use for calculation (needed for terrain), defaults to 0deg
- beamwidth - in degree, to specify the radars beamwidth
- range_res - horizontal counterpart to vert_res
- maxrange - horizontal counterpart to maxalt
- units - default 'm', if 'km' labels and grids adapted to km
- terrain - Show terrain, if True
- cg - Use curvelinear grid, if True
- ax - the axis to plot on, if None, figure and axis will be created on the fly

This seems quite much, but preserves backward compatibility. Anyway I appreciate your comments on the calling signature and other issues with this proposal.

kmuehlbauer, 2020-07-17:
This is an example for a simple scan strategy on cartesian grid without terrain.
[image]

pandasambit15, 2020-07-17:
> ping @pandasambit15 <https://github.com/pandasambit15>

Dear Kai,

Thanks for your response.

This implementation seems quite nice and complete in terms of the necessary
information to be put in the plot. However, I just wanted to know, when the
user chooses to plot terrain (using the corresponding argument), whether
the terrain data (DEM etc.) has to be passed along with the function as an
argument or the used has to prepare it separately, as we were doing in the
beam blockage calculation?

Regards,
Sambit

*Sambit Kumar Panda*
*Scientist/Engineer-SC, ASD/AOSG/EPSA,*
*SAC(ISRO), Ahmedabad*

kmuehlbauer, 2020-07-17:
@pandasambit We have two options here, which can both be implemented side-by-side:

1. use `terrain` keyword to let the user provide an array of dem heights for that particular direction.
1. let wradlib download the necessary SRTM data and do the calculation for the user (this would need internet access and `WRADLIB_DATA` variable to be set to work.

pandasambit15, 2020-07-17:
>    2. let wradlib download the necessary SRTM data and do the calculation

Yes! That seems fine. Whoever doesn't have the internet access should pass
on the required DEM heights as argument. Thanks for the clarification.

Regards,
Sambit

*Sambit Kumar Panda*
*Scientist/Engineer-SC, ASD/AOSG/EPSA,*
*SAC(ISRO), Ahmedabad*

jorahu, 2020-07-17:
Hey

I don't think there are too many parameters. You need all of them to have control over all the drawing process. Just as Mr. Sambit had the issue in the very original post, that "how to adjust the plot". All these parameters give user the control.

Question about the gridline "density". If user defines his/her own axes object, the ticks and ticklabels can be modified outside the plot_scan_strategy function?
Also another issue would be the coloring. There should be option for black and white coloring (maybe some applications might not require colors) (Although I like the colors you have drawn in the example plot above!). Or is it also overwritable by external functions through axes object?

As I wrote when sending the code to you, with curvilinear grid, how is the gridlines drawing managed? The issue I had was that the yticks lables did not match the curvilinear gridlines. This was probably because I did not choose correct axis object to reference the ytick values from (there was this ax, cax and pax axes that I could figure out how to use properly).

And I also love the option for use of terrain data.

Best regards
Jorma Rahu

kmuehlbauer, 2020-07-17:
Here are two images with four panels, describing the four types (cartesian/cg, with/without terrain). I'll come up with a implementation Pull Request next week.

@jorahu To your questions:

- Yes, the curvelinear grid can be customized afterwards to some extent. Synchronizing with the cartesian ticks might not be that straightforward, but manageable. I'll create an example. 
- Yes, we should be able to create a greyscale version of this.

[image]

[image]

pandasambit15, 2020-07-17:
Dear Kai,

Thanks for sharing the plots. These look great.

Regards,
Sambit

kmuehlbauer, 2020-07-22:
@pandasambit15 @jorahu

The updated/fixed `plot_scan_strategy` is available in wradlib 1.7.4. You might test this at your leisure. I'll create an example  notebook which will be available in wradlib-notebooks as well as in the docs. But this might take some time.

pandasambit15, 2020-07-22:
Dear Kai,

Thanks for implementing this in wradlib, that too in such a short time.
Really appreciate it!

Regarding using the new utility, do we have to install the new version
(1.7.4) from the source package or is there any other shortcut/trick to
patch this new function to our existing installations?
The reason why I ask this is because our office machines don't have
internet access (for security reasons), so I usually have to download the
source package and all the dependencies to make thighs work!

Thanks and regards,
Sambit

*Sambit Kumar Panda*
*Scientist/Engineer-SC, ASD/AOSG/EPSA,*
*SAC(ISRO), Ahmedabad*

kmuehlbauer, 2020-07-22:
@pandasambit15  You would need to replace the `vis.py` in your current installation. But you need at least version 1.6 I think, to use the automatic srtm download. This should do the trick.

pandasambit15, 2020-07-22:
Okay! Thanks for the help! Will do the necessary changes and let you know
how it goes.

Regards,
Sambit

*Sambit Kumar Panda*
*Scientist/Engineer-SC, ASD/AOSG/EPSA,*
*SAC(ISRO), Ahmedabad*

jorahu, 2020-07-23:
Hey all

Sorry for being unresponsive. I am actually on vacation at the moment, so
have been a bit distant from computer.

0. I also had a little problems with updating the wradlib version. For some
reason conda update wradlib resulted in memoryerror, but this might be my
anaconda version issue. "pip install wradlib -U" made the trick for me
(without conda update).

Couple of issues I see.
1. The examples Kai provided earlier were for 150 km range I guess. We have
C-band radars, which means we use the 250 km range mostly and when using
the cg grid the lower part (right side) of the plot was eaten away by the
"earth surface" as it went below the ylims. So for longer ranges line 1111
from vis.py ax.set_ylim([ecp.min() - maxalt / 5, ecp.max() + maxalt]) results
in too "high" lower ylim limit. But as the plot_scan_strategy returns ax
then extra lines:
cur_ylim = ax.get_ylim()
ax.set_ylim([cur_ylim[0]-2000, cur_ylim[1]])
gives the control to make everything adjustable again. So I subtracted 2000
meters from the lower ylim just for example.

2. Another problem, when I created figure and axes with
fig = plt.figure(figsize=(10,8))
ax = fig.add_subplot(111)
then while using the cg=True, it resulted in error, that ax object can not
be "AxesSubPlot". I think this is problem around line 1089, as the
greate_cg subplot kwarg requires the number (eg. 111, as default), then
when providing AxesSubPlot object via fig.add_subplot it crashes.

3. With the automatic terrain download I got RuntimeError. Maybe this is
also my system issue, I set the WRADLIB_DATA path to my system, so this
should not be the issue (but I am still not 100% sure). Maybe you already
know, what might cause this error.
Traceback (most recent call last):
  File "<ipython-input-2-eab718e0630b>", line 8, in <module>
    ax = wrl.vis.plot_scan_strategy(ranges, elevs, sitecoords, cg=True,
terrain=True)
  File "C:\Users\IdaJorma\Anaconda3\lib\site-packages\wradlib\vis.py", line
1072, in plot_scan_strategy
    download={},
  File "C:\Users\IdaJorma\Anaconda3\lib\site-packages\wradlib\io\dem.py",
line 148, in get_srtm
    dem = gdal.Warp("", demlist, format="MEM")
  File "C:\Users\IdaJorma\Anaconda3\lib\site-packages\osgeo\gdal.py", line
575, in Warp
    return wrapper_GDALWarpDestName(destNameOrDestDS, srcDSTab, opts,
callback, callback_data)
  File "C:\Users\IdaJorma\Anaconda3\lib\site-packages\osgeo\gdal.py", line
3121, in wrapper_GDALWarpDestName
    return _gdal.wrapper_GDALWarpDestName(*args)
RuntimeError: No usable source images.

4. I couldn't test my own preloaded terrain values because of some shape
issue, but it is getting quite late and I try testing it later some day.
But this part seemed quite easy (lines 1104-1107), when just given the
height values to terrain as np.ndarray. Could just test this with some
random array.

5. But the biggest issue for me was this zigzag pattern.  I have no idea
why this came like this. Also the legend is split for some reason. Could it
be matplotlib version issue? I have '3.1.0' for matplotlib.
[image: image.png]
6. For cg grid the zigzag is also present, but bigger issue is that the
beams are following the ground surface (the examples did not have this
issue), but they should be straight (theoretically) when the cg grid is
applied. This 'straight beam' idea was the whole reason I started messing
around with the cg option from wradlib in the first place. I don't have any
suggestions or solutions here.
[image: image.png]

Units and everything else I tried to test worked wonderfully. I also like
the fact that the overlapping areas can be clearly seen. Also, 4 lines of
code is great improvement from my previous 200 line code :)

Thanks Kai for this upgrade.

Best regards
Jorma Rahu

Kontakt pandasambit15 (<notifications@github.com>) kirjutas kuupäeval K,
22. juuli 2020 kell 14:12:

kmuehlbauer, 2020-07-24:
@jorahu Thanks for the extensive testing (while in vacation). I really appreciate your efforts.

1. **y-axis range**
Yes, I just put in some number there to make this look nice for the test case. Do you think it would be sufficient to add keyword argument `minalt` to set the minimum y-axis range?

2. **AxesSubPlot issue**
There is not much we can do about it, since this uses `mpl_toolkits.axisartist.SubPlotHost`. We could raise an informative error, though. If you need to specify the axis location you can use GridSpec notation. Have a look at https://docs.wradlib.org/en/latest/notebooks/visualisation/wradlib_plot_curvelinear_grids.html, to get an impression (I just see that this is extremely buggy, most likely due to new matplotlib version)

3. **automatic download issue (windows only)**
This is fixed already in latest wradlib master. There was a bug in url creation under windows...

5a. **zigzag pattern**
I think you added an image, but it didn't came through. I do not see a zigzag pattern, but maybe I'm looking at the wrong place.

5b. **legend split**
This was made by purpose, but I'm open to combine this into one legend. 

6. ** beams following ground surface**
I'm a bit confused now. The zero degree beam center follows a line which represents 4/3 * earth radius, thus slightly bent towards earth. But it might be an issue with how the 3db lines are currently calculated. They are just added/subtracted from the calculated center beam height (no refraction included) . Maybe we would need to do this for proper images, and maybe this accounts for the zig/zag (just guessing).

kmuehlbauer, 2020-07-24:
FYI: An example notebook is available [here](https://github.com/wradlib/wradlib-notebooks/blob/devel/notebooks/visualisation/wradlib_plot_scan_strategy.ipynb).

jorahu, 2020-07-24:
Hey

Okay, yes. Thanks for the notebook.

I had my ranges variable with np.arange but as ints (ranges2 = np.arange(0,
250000, 1000)). I guess this resulted in some sort of rounding error,
because when making the sequence as floats (ranges2 = np.arange(0, 250000.,
1000.)), then the plots came out fine, just like the examples were. So this
was my error. Also now with the cg option the beams are straight.
[image: image.png]

Jorma

Kontakt Kai Mühlbauer (<notifications@github.com>) kirjutas kuupäeval R,
24. juuli 2020 kell 14:05:

jorahu, 2020-07-24:
Also the minalt keyword would be sufficient I think to account for the
lower limit. And other issues I described are also covered with correct
ranges definition.
The cg drawing is quite complicated anyway (considering the link you
provided), so I think the subplot issue is minor and the default 111 allows
still very easy drawing for the scan strategies. Not to mention, (nearly)
every aspect can be overwritten by set methods for the resulting ax object
from plot_scan_strategy function.

I try testing the automatic terrain download sometime later. I think when I
updated yesterday, that I got the latest version, but I might be wrong. I
got 1.7.4 version.

I really like the example notebook. It covers all the main features and
possibilities. Very nice work!

Jorma

Kontakt Jorma Rahu (<jorma.rahu@gmail.com>) kirjutas kuupäeval R, 24. juuli
2020 kell 14:20:

kmuehlbauer, 2020-07-24:
> I try testing the automatic terrain download sometime later. I think when I updated yesterday, that I got the latest version, but I might be wrong. I got 1.7.4 version.

The fixes are in the github master. I'm trying to release this as 1.7.5 soonish.

jorahu, 2020-07-24:
Oh, okay. Thanks for the clarification.
I still want to try it out, if it will be released, so looking forward to
it!

Jorma

kmuehlbauer, 2020-07-24:
You can still try instantly:

`pip install git+https://github.com/wradlib/wradlib.git`

will install the latest commit from master branch.

kmuehlbauer, 2020-08-31:
resolved by #442

PR #442 (2020-07-21): FIX: update `vis.plot_scan_strategy()`

closes #441
closed 2020-08-31 by kmuehlbauer