wradlib/wradlib #444: Documentation in vpr.py is not consistent

fuxb, 2020-07-23:
In vpr.py, documentation of parameters of classes that use `class CartesianVolume` i.e. `class CAPPI(CartesianVolume)` and `class PseudoCAPPI` is not consistent with parameters of `class CartesianVolume` and thus is wrong also on docs.wradlib.org.

More precisely, description of `gridshape` is inconsistent, and `minelev` and `maxelev` are only described in `CartesianVolume` but not in `PseudoCAPPI` and `CAPPI`.

kmuehlbauer, 2020-07-23:
@fubx Thanks for bringing this to attention. 

It seems that `gridshape` isn't used at all in these functions. Good chance that the output should've been reshaped to this `gridshape`. 

So, this will need some rework, possible with breaking changes involved.

@fubx Do you feel like jumping in to resolve this?

kmuehlbauer, 2020-07-23:
@fuxb Thanks for bringing this to attention.

It seems that gridshape isn't used at all in these functions. Good chance that the output should've been reshaped to this gridshape.

So, this will need some rework, possible with breaking changes involved.

@fuxb Do you feel like jumping in to resolve this?

fuxb, 2020-07-24:
@kmuehlbauer  Unfortunately I don't have time to rewrite the code. But if you think it will suffice for now, I can just update the documentation so that at least there is a comment that `gridshape` is not used. And I would include `minelev` and `maxelev` in the description.

kmuehlbauer, 2020-07-24:
@fuxb Yes, fixing the documentation should suffice for now. :+1: 

There need to be a discussion about API anyway, if this will be rweorked. There might be one or another minor release between.

PR #505 (2021-08-16): FIX: remove unnecessary `gridshape` kwarg from docstring in ``CartesianVolume``
- closes: #444 
- fix docstrings
- minor fixes
closed 2021-08-16 by kmuehlbauer