wradlib/wradlib #448: DWD WN RADOLAN Grid 1200x1100

julste, 2020-08-29:
Hi there!

Since the FX radar product isn't available on September 30, 2020 anymore but its successor product WN radar product ([https://www.dwd.de/DE/leistungen/opendata/neuigkeiten/opendata_jul2020_01.html](https://www.dwd.de/DE/leistungen/opendata/neuigkeiten/opendata_jul2020_01.html)), is it possible to support a 1200 x 1100 Grid? Because the WN-Composit is using 1200 x 1100 as you can see [here](https://www.dwd.de/DE/leistungen/radarprodukte/radarkomposit_wn.pdf?__blob=publicationFile&v=3). 

And is there a workaround so far?

I tried other dimensions like the extended one, but their are incompatible:

`TypeError: Dimensions of C (1200, 1100) are incompatible with X (1400) and/or Y (1500); see help(pcolormesh)
`

kmuehlbauer, 2020-08-29:
@julste Thanks for bringing this to attention. I think it should be possible to implement and cut a release before September 30th.

kmuehlbauer, 2020-08-31:
@julste The 1200,1000 grid is implemented in #449. You can install wradlib from github master to test. This will be available in the next release.

PR #449 (2020-08-31): ENH: add WN product size (1200,1000) to radolan grid
This adds WN product to `georef.get_radolan_grid()`.

Please note, that the grid aligns with the other DWD (RADOLAN) grids with the origin in the lower left point (not pixel center). Also note that the WN product can be read with `io.read_radolan_composite()` which returns data in the same fashion (origin is lower left).

fixes: #448
closed 2020-08-31 by kmuehlbauer