wradlib/wradlib #453: Feature Request: UF reader

kayarre, 2020-09-04:
I would like to implement a uF reader in wradlib, but its unclear to me where that functionality would go, and where I should start.

pyart for example has a uf reader, [uf reader](https://github.com/ARM-DOE/pyart/blob/master/pyart/io/uf.py)

and Provides a generic class `Radar` that can be used to create the data object, but it looks like wradlib is tailored to the more modern netcdf/hdf5 based data stores, which have the data structure defined when the file is read.

I have parsed the binary UF files into a data structure, but I need to populate that data into xarray CfRadial format for `wradlib` but it is not clear **to me** how to build the data structure with `xarray` in  [xarray.py](https://github.com/wradlib/wradlib/blob/master/wradlib/io/xarray.py)

my first thought was to take an CfRadial file and somehow export the schema, and then use that as a basis for populating the structure from the UF file format. 

I would appreciate any input from the community.

kmuehlbauer, 2020-09-05:
@kayarre  First, welcome to wradlib. Capability of a uF reader has been asked every once in a while, we've recommended pyart for this in the past.

Tldr: Yes, this would be a great addition! 

But in the light of the recent developments in the world of standards (CfRadial2) and coding (xarray) a uF reader would fit nicely in wradlib.

Most current readers just read the data from their respective raw source files into a nested dictionary of numpy arrays (and other dtypes).

With the xarray based readers for CfRadial and ODIM  standard data files we tried to adopt CfRadial2 standard using xarray. As you might have seen in the recent commit history the development is not yet finished, as needed features will be added.

The idea is to add this xarray CfRadial structure to other wradlib readers (Rainbow and Sigmet/Iris), see wradlib issues.

This should be finalised in wradlib release 2.0, which has no due date yet. At least the overall xarray CfRadial implementation should be mature by then. 

I've never touched uF files myself, so I've no idea how this data would be filled into the xarray structure. 

The latest implementation in wradlib is going ahead of the current proposed CfRadial2 standard, by allowing timeseries of sweeps. There is [this notebook](https://docs.wradlib.org/en/stable/notebooks/fileio/wradlib_odim_multi_file_dataset.html) , where the functionality is written up. Maybe this can help you for the start. 

Looking forward to assist in this endeavour. :+1:

kmuehlbauer, 2020-09-05:
@kayarre I've just looked a bit at the UF specs. We might be able to borrow bits of code from the current sigmet/iris reader. But that depends on your current implementation. The conversion to the xarray/CfRadial2 structure would be a blueprint for the other readers(sigmet/iris, rainbow...). So we could join forces there.

I know it might get hard to find the way through the organic grown wradlib code. We have code that hasn't been touched much since the project went online in 2012. OTOH we've bleeding edge code like the xarray-ODIM/CfRadial2 implementation. And we try to keep pace with the developments in the community (eg. using conda-forge, continuous integration, xarray etc.). 

Maybe it would be a good idea to schedule an online meeting (eg. via zoom) to answer questions (yours and mine) after you made your first steps with the code?

kayarre, 2020-09-07:
Thank you @kmuehlbauer, leveraging the current the `xarray-ODIM/CfRadial2 implementation` is what I was planning to target.

Currently there are few paths I could follow. 1. use pyart, 2. leverage part the pyart uf reader to make a new one in wradlib within the xarray development, or 3. use the uf parser I made to to integrate with xarray/wradlib. (I didn't know UF was an informally standard file type and used a provided matlab uf reader creating a binary reader from a yaml file using  [kaitai struct](https://kaitai.io/). It is nothing special, but useful)

I think part of the desire is to limit the number of dependencies while getting nice features.  e.g. leverage xarray with the CfRadial2 format only depending on wradlib. 

Yes I would be happy to speak with you in an online meeting. I will try to make some progress today and see where it leads.

kmuehlbauer, 2022-04-03:
Just as a reminder, the UF format is described in all detail within the `IRIS Programming guide M211318EN-H` in Appendix C (pp. 127-131). So whoever is trying to get a start here should have a look there first and the implementation of the IRIS reader.

kmuehlbauer, 2022-11-07:
This will be tackled in the new package [xradar](https://github.com/openradar/xradar) which wradlib will depend on in the future. Please open new discussion topics over there.

closed 2022-11-07 by kmuehlbauer