wradlib/wradlib #454: Convert xy-space to lonlat-space using WN-Komposit

julste, 2020-09-06:
Hello @kmuehlbauer,

this isn't a bug report but more a question about how to convert xy-space to lonlat-space using WN data from DWD:

I downloaded the WN000 latest data, unpacked it and tried to plot it:
```
dwd_string = "+proj=stere +lat_0=90 +lat_ts=60 +lon_0=10 +a=6370040 +b=6370040 +no_defs +y_0=3609144.7242655749 +x_0=542962.16692185635"

proj_stereo = wradlib.georef.proj4_to_osr(dwd_string)
proj_wgs = osr.SpatialReference()
proj_wgs.ImportFromEPSG(4326)

radolan_wnrid_xy = wradlib.georef.get_radolan_grid(1200, 1100)
x1 = radolan_wnrid_xy[:, :, 0]
y1 = radolan_wnrid_xy[:, :, 1]

radolan_grid_ll = wradlib.georef.reproject(radolan_wnrid_xy, projection_source=proj_stereo, projection_target=proj_wgs)
lon1 = radolan_grid_ll[:, :, 0]
lat1 = radolan_grid_ll[:, :, 1]

fig = pl.figure(figsize=(10, 8))
ax = fig.add_subplot(111, aspect="equal")
pm = ax.pcolormesh(lon1, lat1, data, cmap="viridis")
pl.xlim((lon1[0, 0], lon1[-1, -1]))
pl.ylim((lat1[0, 0], lat1[-1, -1]))
pl.xlabel("Longitude ")
pl.ylabel("Latitude")
```
As a result I got this:
[image]

The long / lat axis seems to be wrong, because shouldn't be it look more like this:

[image]

I'm wondering why the proj-String in the WN-Komposit is [different](https://www.dwd.de/DE/leistungen/radarprodukte/formatbeschreibung_wndaten.pdf?__blob=publicationFile&v=4) and how I can use it.

At the end I would like to translate it to a mbtile file using the following Worldfile:
```
1000
0.0
0.0
-1000
-543.4622
-3609.645
```

Do you have any suggestions for me?

I appreciate any kind of help!

kmuehlbauer, 2020-09-07:
@julste We use the description in [RADOLAN Kompositformat](https://www.dwd.de/DE/leistungen/radolan/radolan_info/radolan_radvor_op_komposit_format_pdf.pdf;jsessionid=941A9B70081C817216C8F1A9207F49CE.live11044?__blob=publicationFile&v=13) to derive the RADOLAN coordinates. 

```python
# get RADOLAN coordinates 
rado_rad = wrl.georef.get_radolan_grid(1200, 1100)

# get WGS84 coordinates derived via reprojection using gdal osr
rado_wgs_proj = wrl.georef.get_radolan_grid(1200, 1100, wgs84=True)

# get WGS84 coordinates derived by pure trigonometry (it's a sphere)
rado_wgs_trig = wrl.georef.get_radolan_grid(1200, 1100, wgs84=True, trig=True)

# check for equality
np.testing.assert_array_almost_equal(rado_wgs_trig, rado_wgs_proj)

```
I've really no idea why the DWD has this other proj-string in the WN product. But you can adapt it to use with your example like this (remove `x_0` and `y_0` and add `+units=km`):

`dwd_string = "+proj=stere +lat_0=90 +lat_ts=60 +lon_0=10 +a=6370040 +b=6370040 +no_defs +units=km"`

HTH!

julste, 2020-09-07:
Thank you for your support.The axes seem to be correct now, so the plot looks good to me :+1: 

Unfortunately the wrapping (EPSG:3857) and the translation to the mbtile haven't worked so far, because it ended in a wrong area. 

But that seems to be am issue with the worldfile.

Thanks again!

kmuehlbauer, 2020-09-08:
@julste You can use wradlib built-in gdal machinery to do what you want:

```python
# create projection
proj_3857 = wrl.georef.epsg_to_osr(3857)

# bring data/coords in upper-left origin
data, radolan_wngrid_xy = wrl.georef.set_raster_origin(data, radolan_wngrid_xy, 'upper')

# create in-memory dataset
ds = wrl.georef.create_raster_dataset(data, radolan_wngrid_xy, projection=proj_stereo)

# reproject/resample to target projection
ds1 = wrl.georef.reproject_raster_dataset(ds, projection_target=proj_3857, spacing=1000)

# output to MBTiles
wrl.io.write_raster_dataset("mytiles.mbtiles", ds1, 'MBTiles')
```

While reprojecting you can also resample (kwargs `spacing` vs. `size`, `resample`) and align the new grid (kwarg `align`).
For the output to `MBTiles` (you can output to any gdal raster format here) you can set options to specify the tiles (see [gdal MBTiles driver](https://gdal.org/drivers/raster/mbtiles.html#mbtiles))

julste, 2020-09-08:
I'm sorry for bothering you again, but I don't get it.

If I transfrom the data set to MBtiles as the way you described it without plotting it before, how can I e.g. set the color of the pixel? 

My MBtiles file with plotting (wrong area) has a size of around 50 mb, the other MBtiles file (built-in gdal machinery) without plotting only around 16 kb. So I think there is some information missing.

Do you have an example how to create a colored MBtiles file?

Thanks again!

kmuehlbauer, 2020-09-09:
Now you got me :grimacing:. I'm no GIS expert, so bear with me.

If you want to go the gdal road (maybe with use of wradlib) you also want to read [gdal doc](https://gdal.org/drivers/raster/mbtiles.html?highlight=mbtiles) concerning MBTile creation options and handling.

1. create gdal color table from matplotlib colormap
    ```python
    from osgeo import gdal
    ct = gdal.ColorTable(gdal.GPI_RGB)
    for i, c in enumerate(ct_viridis):
        ct.SetColorEntry(i, tuple(c))
    ```
2. add the following for the raster creation
    ```python
    # reproject/resample to target projection
    ds1 = wrl.georef.reproject_raster_dataset(ds, projection_target=proj_3857, spacing=1000)
    band = ds1.GetRasterBand(1)
    band.SetRasterColorTable(ct)
    band.SetRasterColorInterpretation(gdal.GCI_PaletteIndex)
    ````
That will create a MBTiles file with `gdalinfo`:

```
Driver: MBTiles/MBTiles
Files: mytiles.mbtiles
Size is 1380, 1473
Coordinate System is:
PROJCRS["WGS 84 / Pseudo-Mercator",
    BASEGEOGCRS["WGS 84",
        DATUM["World Geodetic System 1984",
            ELLIPSOID["WGS 84",6378137,298.257223563,
                LENGTHUNIT["metre",1]]],
        PRIMEM["Greenwich",0,
            ANGLEUNIT["degree",0.0174532925199433]],
        ID["EPSG",4326]],
    CONVERSION["unnamed",
        METHOD["Popular Visualisation Pseudo Mercator",
            ID["EPSG",1024]],
        PARAMETER["Latitude of natural origin",0,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8801]],
        PARAMETER["Longitude of natural origin",0,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8802]],
        PARAMETER["False easting",0,
            LENGTHUNIT["metre",1],
            ID["EPSG",8806]],
        PARAMETER["False northing",0,
            LENGTHUNIT["metre",1],
            ID["EPSG",8807]]],
    CS[Cartesian,2],
        AXIS["x",east,
            ORDER[1],
            LENGTHUNIT["metre",1]],
        AXIS["y",north,
            ORDER[2],
            LENGTHUNIT["metre",1]],
    ID["EPSG",3857]]
Data axis to CRS axis mapping: 1,2
Origin = (159077.734018745861249,7530760.759670206345618)
Pixel Size = (1222.992452562820063,-1222.992452562820063)
Metadata:
  ZOOM_LEVEL=7
  name=mytiles
  type=overlay
  description=mytiles
  version=1.1
  format=png
  bounds=1.42901959832139358,45.680479661900776,16.5901524108213962,55.8608308918294227
  minzoom=7
  maxzoom=7
Image Structure Metadata:
  INTERLEAVE=PIXEL
Corner Coordinates:
Upper Left  (  159077.734, 7530760.760) (  1d25'44.47"E, 55d51'38.99"N)
Lower Left  (  159077.734, 5729292.877) (  1d25'44.47"E, 45d40'49.73"N)
Upper Right ( 1846807.319, 7530760.760) ( 16d35'24.55"E, 55d51'38.99"N)
Lower Right ( 1846807.319, 5729292.877) ( 16d35'24.55"E, 45d40'49.73"N)
Center      ( 1002942.526, 6630026.818) (  9d 0'34.51"E, 51d 2'57.64"N)
Band 1 Block=256x256 Type=Byte, ColorInterp=Red
  Mask Flags: PER_DATASET ALPHA 
Band 2 Block=256x256 Type=Byte, ColorInterp=Green
  Mask Flags: PER_DATASET ALPHA 
Band 3 Block=256x256 Type=Byte, ColorInterp=Blue
  Mask Flags: PER_DATASET ALPHA 
Band 4 Block=256x256 Type=Byte, ColorInterp=Alpha
```
It has now RGBA Bands and can be visualised nicely in QGIS. You might need to create the other zoomlevels with `gdaladdo` as hinted in the above gdal link.

julste, 2020-09-09:
Thank you very much, it helped me a lot!

kmuehlbauer, 2020-09-10:
@julste Great! Would you mind sharing your final code? I think it would fit nicely into wradlib-notebooks as example of GIS export.

julste, 2020-09-10:
I haven't figured out a working solution yet, need some time to understand the concept better. Your answers will help me to do that, but I don't want to waste your time anymore. 

As soon as got a working example (with different zoom levels, colored pixels, center in Germany, bigger file size, running on my tile server), I will share my code here.

Thanks again!

kmuehlbauer, 2020-09-10:
@julste  No worries, take your time.

julste, 2020-10-10:
@kmuehlbauer I have decided to use WMS tiles from DWD geoserver, so I won't implement it myself anymore. 

Thanks again for your support.

kmuehlbauer, 2020-10-12:
@julste Thanks for coming back to this. I'll close this because the original issue is resolved. Please reopen, if needed.

closed 2020-10-12 by kmuehlbauer