wradlib/wradlib #456: Can't install wradlib in development

zssherman, 2020-09-19:
Hello Wradlib team! 

When I try to install wradlib in development mode (reason being trying to see if there is possibly an issue in the cumulative beam block code, if so want to maybe find a fix, might not be an issue at all, not sure yet) I get this error when I do 'pip install -e .' on the clone or python setup install I get:

ERROR: Command errored out with exit status 1:
   command: /home/zsherman/anaconda3/envs/cmac_env/bin/python3.7 /home/zsherman/anaconda3/envs/cmac_env/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpny670fdj
       cwd: /home/zsherman/dev/wradlib
  Complete output (31 lines):
  fatal: No tags can describe '1f449ea65aa96f1e2bd056b6427825fdbe4035f3'.
  Try --always, or create some tags.
  WRADLIB: Unable to import git_revision from repository.
  Traceback (most recent call last):
    File "/home/zsherman/anaconda3/envs/cmac_env/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/home/zsherman/anaconda3/envs/cmac_env/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/home/zsherman/anaconda3/envs/cmac_env/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 114, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/tmp/pip-build-env-z4nc8hrk/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 150, in get_requires_for_build_wheel
      config_settings, requirements=['wheel'])
    File "/tmp/pip-build-env-z4nc8hrk/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 130, in _get_build_requires
      self.run_setup()
    File "/tmp/pip-build-env-z4nc8hrk/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 254, in run_setup
      self).run_setup(setup_script=setup_script)
    File "/tmp/pip-build-env-z4nc8hrk/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 145, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 199, in <module>
      setup_package()
    File "setup.py", line 168, in setup_package
      VERSION = write_version_py()
    File "setup.py", line 116, in write_version_py
      GIT_REVISION, GIT_HASH = git_version()
    File "setup.py", line 60, in git_version
      git_rev = check_output(["git", "describe", "--tags", "--long"])
    File "/home/zsherman/anaconda3/envs/cmac_env/lib/python3.7/subprocess.py", line 411, in check_output
      **kwargs).stdout
    File "/home/zsherman/anaconda3/envs/cmac_env/lib/python3.7/subprocess.py", line 512, in run
      output=stdout, stderr=stderr)
  subprocess.CalledProcessError: Command '['git', 'describe', '--tags', '--long']' returned non-zero exit status 128.
  ----------------------------------------
ERROR: Command errored out with exit status 1: /home/zsherman/anaconda3/envs/cmac_env/bin/python3.7 /home/zsherman/anaconda3/envs/cmac_env/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpny670fdj Check the logs for full command output.

Not sure of any way around this, I would temporary just put an arbitrary version, but seems version is imported throughout wradlib. Thanks for your time!

kmuehlbauer, 2020-09-19:
@zssherman Unfortunately I can't reproduce this on a clean conda-forge environment and a freshly cloned wradlib-repo. The error `fatal: No tags can describe '1f449ea65aa96f1e2bd056b6427825fdbe4035f3'.` indicates that something is wrong with the git `tag` history. 

There is a [SO question](https://stackoverflow.com/questions/6445148/git-fatalno-tags-can-describe-sha1-number/41605474) where the recommendation is to fetch from the remote. Another problem might be a shallow clone.

If this doesn't help please provide more details how to reproduce. Thanks!

zssherman, 2020-09-19:
Hi @kmuehlbauer! A new clone fixed my issue! Thanks for your help! Closing.

closed 2020-09-19 by zssherman

kmuehlbauer, 2020-09-19:
@zssherman Glad it worked!