wradlib/wradlib #459: Support of input of EEC Edge and Rainbow volume data with conversion to HDF5

maxok, 2020-09-24:
Is there any plan for implementation of support to input EEC Edge and Gematronik Rainbow Polar Volume Data for conversion to HDF5?

kmuehlbauer, 2020-09-25:
@maxok Thanks for asking. Currently we have the [EDGE netcdf reader](https://docs.wradlib.org/en/stable/notebooks/fileio/wradlib_radar_formats.html#EDGE-NetCDF) and the [Gematronik Rainbow reader](https://docs.wradlib.org/en/stable/notebooks/fileio/wradlib_radar_formats.html#Gematronik-Rainbow). Both can read polar volumes. 

Do those readers not fit your requirements? If so, what should be changed or improved? How should the conversion to hdf5 lokk like? Should it convert to some format like ODIM_H5? Please give more detail for this.

maxok, 2020-09-25:
@kmuehlbauer Thank you very much for your quick response. You are totally correct. I was asking about functionality to convert EDGE netcdf and Gematronik Rainbow to ODIM_H5.

Many thanks in advance.

kmuehlbauer, 2020-09-25:
@maxok OK, for the Rainbow reader this is tracked by #394. There is no issue for the Edge Netcdf reader so far, but we could just rename this one.

kmuehlbauer, 2020-09-25:
@maxok We are heading for an `xarray` based implementation of `CfRadial2` matching standard. Using this we would be able to export to `CfRadial2` and `ODIM_H5`. For the EEC EDGE NetCDF data it should not be problematic, since the source files can be opened with `xarray` quite easy, we would need a bit of boilerplate code to align it with the current implementation details (see https://docs.wradlib.org/en/stable/notebooks/fileio/wradlib_odim_multi_file_dataset.html for the class structure so far).

Any help is much appreciated. I'll tag this with the `Hacktoberfest` label.

kmuehlbauer, 2021-10-05:
@maxok For the rainbow5 reader please have a look at: #522