wradlib/wradlib #461: Feature: 3D data grid output from polar volume data

meteoDaniel, 2020-10-02:
Dear guys, 
I think analyzing 2D radar data is a basic tool now in many scientific fields. But the standard in weather radar data processing is to scan more or less the whole sphere around a radar station but there are no tools to generate nice data fields from it. 
So my feature request is a 3D xarray with reflectivity or other parameters and corresponding latitude, longitude and altitude coordinates. 
Afterwards I will provide a recipe plotting such a data with matplotlib and plotly. 
Best regards

kmuehlbauer, 2020-10-02:
@meteoDaniel Thanks for raising this.

Currently we have some tools in the `vpr` module to create a cartesian volume. See also [recipe 2 in the docs ](https://docs.wradlib.org/en/stable/notebooks/workflow/recipe2.html). Basically we could use that as a starting point.

For `xarray` we are on the way to create a container to hold radar volume/timeseries according to CfRadial2/ODIM_H5 standards. You can start with DWD radar data and `wradlib.io.open_odim` to load data into this container. 

Let me know if you will have time to look into this. I'll tag this with hacktoberfest.

kmuehlbauer, 2023-09-07:
Radar Volume reading is able with xradar-package. 3D Grid interpolation might be possible using ipol-methods. But nevertheless it might be better to rely on 3rd party packages which are dedicated to these interpolation/gridding tasks. Closing for now. Feel free to reopen or open a new issue.

closed 2023-09-07 by kmuehlbauer