wradlib/wradlib #466: CfRadial2 issues

kurtsansom, 2020-11-05:
I am trying to read and plot a quasi-compliant CfRadial2 file.
It appears to be choking on because variables and attributes are missing from the data, but aren't supposed to be specified in the variable.

I still need to checkout the wradlib test data to see what the differences are other than the issue I am seeing.

#### MCVE Code Sample

```python
import wradlib as wrl
data = wrl.io.xarray.CfRadial(filename=file_new, flavour='Cf/Radial2')

data[data.sweep_names[0]].dbz.attrs['sweep_mode'] = data[data.sweep_names[0]].sweep_mode

# I also tried adding new dimensions of bins, but it didn't make a difference  e.g.
# new_data = data[data.sweep_names[0]].assign_coords(bins=("bins", **data[data.sweep_names[0]].coords['range']))

fig = plt.figure(figsize=(10,10))
data[data.sweep_names[0]].dbz.sortby('azimuth').wradlib.plot_ppi(proj='cg', fig=fig)

```

#### Expected Output
A nice pretty picture of the radar data.

#### Problem Description
Currently when `data[data.sweep_names[0]].dbz` is passed to plot_ppi, it expects the `sweep_mode` attribute to exist,  and after I added that one to the dataset it was still missing the `bins` array.
It appears in the CfRadial2 document that maybe it should be `range` instead?

This may be an issue with the Cfradial convention being incomplete or some other issue?

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

INSTALLED VERSIONS
------------------
commit: None
python: 3.8.6 | packaged by conda-forge | (default, Oct  7 2020, 19:08:05) 
[GCC 7.5.0]
python-bits: 64
OS: Linux
OS-release: 5.4.0-52-generic
machine: x86_64
processor: x86_64
byteorder: little
LC_ALL: None
LANG: en_US.UTF-8
LOCALE: en_US.UTF-8
libhdf5: 1.10.6
libnetcdf: 4.7.4

xarray: 0.16.1
pandas: 1.1.3
numpy: 1.19.2
scipy: 1.5.2
netCDF4: 1.5.4
pydap: None
h5netcdf: 0.8.1
h5py: 2.10.0
Nio: None
zarr: None
cftime: 1.2.1
nc_time_axis: None
PseudoNetCDF: None
rasterio: None
cfgrib: None
iris: None
bottleneck: None
dask: None
distributed: None
matplotlib: 3.3.2
cartopy: 0.18.0
seaborn: None
numbagg: None
pint: None
setuptools: 50.3.0.post20201103
pip: 20.2.4
conda: None
pytest: None
IPython: 7.19.0
sphinx: None

wradlib: 1.8.1-dev0+gb4b91e0

</details>

#### additional comments
I was able to work through some of the issues with the input format. e.g. renaming `receiver_latitude` to `latitude` using the netcdf-python bindings because Xarray cannot handle groups itself.

kmuehlbauer, 2020-11-06:
@kurtsansom Thanks for raising this.

The current CfRadial2 implementation is still "in the works", since the standard is not yet approved. Here we try to hold to the standard as laid out in https://github.com/NCAR/CfRadial.

You can add the georeferencing coordinates like this:

`data = data.pipe(wrl.georef.georeference_dataset)`

This will add 2D coordinates `x`, `y`, `z` and `gr` (ground range) in azimuthal equidistant radar centric projection as well as `bins` and `rays` for the `cg` plotting capabilities. 

Can you make available one of these files? I might have a look for possible faults in the current implementation.

kurtsansom, 2020-11-06:
thank you @kmuehlbauer,
I have been working through the CfRadial 2.1 document as well and I will see if I can get permission to share the file. 

So it appears the issue with the `bins` was resolved with adding the georeferencing.
```python
data = wrl.io.xarray.CfRadial(filename=file_new, flavour='Cf/Radial2', georef=True)
```

That still didn't resolve the issue with the sweep mode, but I was able to circumvent that issue with 
`data[data.sweep_names[0]].dbz.attrs['sweep_mode'] = data[data.sweep_names[0]].sweep_mode`

This may be a non-issue now as well as the format of the files being used on my end may not be correct either.

kurtsansom, 2020-11-06:
Should I make a pull request to the data repo or do recommend another avenue to pass on sample data?

kmuehlbauer, 2020-11-06:
@kurtsansom Thanks for coming back! A PR to wradlib-data would be fine. That way we can evaluate before final merge. Please also add a short description of the provenience. I'll have a look as soon as possible, but might get to it only after the weekend.