wradlib/wradlib #473: Need Iris RAW data plotting

mustafaalicelik, 2021-01-01:
Here is an example for Vaisala Sigmet RAW data but there is no plotting for that:
https://docs.wradlib.org/en/stable/notebooks/fileio/wradlib_radar_formats.html
How can I continue with that example and get a plot for my RAW data?

https://gist.github.com/mustafaalicelik/91cf2cd022694296296fbb7077a7cc6f

kmuehlbauer, 2021-01-01:
It looks like enhancing the linked documentation with a plotting example would be good to give users an idea how to proceed. I'll have a look into [the example](https://github.com/wradlib/wradlib-notebooks/blob/master/notebooks/fileio/wradlib_radar_formats.ipynb) next week, if no one beats me to it.

@mustafaalicelik In your gist you are already on a good way, extracting the needed values. You would need to setup range,  azimuth and site coords and feed the arrays to [`wradlib.vis.plot_ppi`](https://docs.wradlib.org/en/stable/generated/wradlib.vis.plot_ppi.html).

mustafaalicelik, 2021-01-01:
Thanks for interest. Hope I'll find out as quick as possible.

kmuehlbauer, 2021-01-08:
@mustafaalicelik Please have a look in the updated example [1]. It's very simple, but it should get you along.

[1] https://docs.wradlib.org/en/latest/notebooks/fileio/wradlib_radar_formats.html#Vaisala-Sigmet-IRIS

mustafaalicelik, 2021-01-08:
I'm grateful, it works perfectly.

Kai Mühlbauer <notifications@github.com>, 8 Oca 2021 Cum, 18:11 tarihinde
şunu yazdı:

kmuehlbauer, 2021-01-08:
@mustafaalicelik OK, I'm going to close the issue. If you encounter any problems you might open a new issue.

closed 2021-01-08 by kmuehlbauer