wradlib/wradlib #482: Incompatible to latest xarray (0.17.0)

s-m-e, 2021-03-01:
#### MCVE Code Sample

```python
import wradlib as wrl
```

#### Expected Output

None.

#### Problem Description

An exception occurs:

```
cannot import name '_MultiFileCloser' from 'xarray.backends.api'
```

#### Version

1.9.0

#### Workaround

Downgrade xarray to 0.16.

kmuehlbauer, 2021-03-01:
@s-m-e Please install from github master until wradlib 1.10 is released.

four43, 2021-03-01:
Conda users - another quick solution is to manually specify xarray and the desired version constraint:
```
conda install --override-channels --channel conda-forge 'wradlib' 'xarray<0.17'
```

kmuehlbauer, 2021-03-01:
Thanks @four43! Correct solution.

kmuehlbauer, 2021-03-02:
@s-m-e @four43 I've released wradlib 1.9.1 just now, to fix this issue.  conda-forge package should be available in some hours.

wradlib 1.10 needs some more time.

closed 2021-03-02 by kmuehlbauer

four43, 2021-03-02:
Thanks for the quick update, Kai. It's appreciated!
-Seth