wradlib/wradlib #483: Missing documentation? Plotting a (RADOLAN) composit with cartopy

s-m-e, 2021-03-01:
I am just trying to figure this out - I am somehow failing to plot a RADOLAN composite with `cartopy`. I am not getting the (conversion of the) projection right and the plot is "empty". This is a minimal example:

```python
from io import BytesIO
import requests

import numpy as np

from pyproj import CRS

import wradlib as wrl

import cartopy.crs as ccrs
import cartopy.feature as cfeature
import matplotlib.pyplot as plt

from salem.gis import proj_to_cartopy

URL = "http://opendata.dwd.de/weather/radar/composit/wx/raa01-wx_10000-latest-dwd---bin"

raw = requests.get(URL).content

data, _ = wrl.io.radolan.read_radolan_composite(BytesIO(raw))
data[data < 0] = 0 # "cleanup"

radolan_grid_xy = wrl.georef.get_radolan_grid(1100, 900)
data_extent = [
    radolan_grid_xy[..., 0].min(), radolan_grid_xy[..., 0].max(),
    radolan_grid_xy[..., 1].min(), radolan_grid_xy[..., 1].max(),
]

# <black_magic>
radolan = str(wrl.georef.create_osr("dwd-radolan"))
radolan_proj = CRS.from_string(osr)
radolan_cartopy = proj_to_cartopy(radolan_proj)
# </black_magic>

fig = plt.figure()
ax = fig.add_subplot(1, 1, 1, projection = ccrs.PlateCarree())
ax.coastlines()
test = ax.imshow(
    data,
    transform = radolan_cartopy,
    extent = data_extent,
    origin = 'upper',
    cmap = 'gray',
    vmin = 0,
    vmax = 255,
)
```

If something like this was in the documentation, I'd appreciate it a lot. The part tagged "black magic" is main main suspect. There must be a better way ... Any ideas?

kmuehlbauer, 2021-03-01:
@s-m-e Thanks for raising this. In #480 I've revamped the RADOLAN reader and with xarray 0.17.0 you can directly import RADOLAN data as xarray Dataset including the coordinates (but without attached CRS for the moment). This will be available in forthcoming wradlib 1.10.

I've planned to rework the documentation also in this regard. I'd appreciate any help with this. Thanks for putting this together.

s-m-e, 2021-03-01:
This sounds promising. Nevertheless, any idea how to get the CRS right? In theory, this should be a straight-forward process ...

s-m-e, 2021-03-01:
I should add that the "scope is clear" - no rain today. I have attached a file from last week. It contains "a few" more features, though the plot remains empty.

[raa01-wx_10000-2102261215-dwd---bin.zip](https://github.com/wradlib/wradlib/files/6061685/raa01-wx_10000-2102261215-dwd---bin.zip)

kmuehlbauer, 2021-03-01:
@s-m-e I've tinkered with cartopy's Stereographic and NorthPoleStereo projections with no success so far. This is an unresolved issue also on my side.

s-m-e, 2021-03-01:
Thanks for trying and letting me know!

kmuehlbauer, 2021-06-01:
@s-m-e I think I've found the problem source (units m vs km). 

I've added a chapter to the next docs: https://docs.wradlib.org/en/latest/notebooks/radolan/radolan_quickstart.html#Map-plot-using-cartopy

I'll close this one. Please reopen, if there are more questions.

closed 2021-06-01 by kmuehlbauer

s-m-e, 2022-10-31:
Thanks for the added documentation.

Just came across the problem once more and wanted to plot this data more or less independently from `xarray`, i.e. directly from a `numpy` array with `matplotlib` and `cartopy` only. There is a solution which appears to work as of writing this:

```python
import cartopy.crs as ccrs
import cartopy.feature as cfeature
import matplotlib.pyplot as plt
from pyproj import CRS
import wradlib as wrl

_grid = wrl.georef.get_radolan_grid(1100, 900)
EXTEND = [
    _grid[..., 0].min(), _grid[..., 0].max(),
    _grid[..., 1].min(), _grid[..., 1].max(),
]
del _grid

TRANSFORM = ccrs.Projection(CRS.from_string(wrl.georef.create_osr("dwd-radolan").ExportToProj4()).to_dict())
TRANSFORM.bounds = EXTEND  # HACK

data, _ = wrl.io.radolan.read_radolan_composite(filename)  # TODO set `filename` to something meaningful ...

fig = plt.figure()
ax = fig.add_subplot(
    1, 1, 1,
    projection = ccrs.EuroPP(),  # just an example, does not matter
    extent = (4, 16, 46, 56),  # lon,lat
)

ax.set_facecolor('#000000')
ax.add_feature(cfeature.BORDERS, color='#808080')

ax.imshow(
    data,
    transform = TRANSFORM,
    extent = EXTEND,
    origin = 'upper',
    cmap = 'jet',
    vmin = 0,
    vmax = 255,
)

ax.gridlines(draw_labels=True, dms=True, x_inline=False, y_inline=False, color='#FFFFFF')
```

The `TRANSFORM` object is a bit "hacky" but essentially what I was looking for. I am just guessing that there are better ways ...

kmuehlbauer, 2022-11-03:
@s-m-e Nice writeup, thanks for sharing.