wradlib/wradlib #484: Cannot find relevant SRTM files for site_coordinates in NW region

gewitterblitz, 2021-03-03:
Hi,

I was going through the tutorial notebook at this [link](https://github.com/wradlib/wradlib-notebooks/blob/8e3ff5df06479ef7a372ecaabc6b34a6c0d5b079/notebooks/visualisation/wradlib_plot_scan_strategy.ipynb) when I spotted a bug in the `wradlib.io.get_srtm` function. I wanted to use the `plot_scan_strategy` function with terrain using site coordinates for a radar located in Alabama, US. However, wradlib did not like it. Sample code shown below:

```
import wradlib as wrl
import matplotlib.pyplot as pl
import matplotlib as mpl
import os,glob
import warnings
warnings.filterwarnings('ignore')
try:
    get_ipython().magic("matplotlib inline")
except:
    pl.ion()
import numpy as np

nrays = 360
nbins = 150
range_res = 1000.
ranges = np.arange(nbins) * range_res
elevs = [ 0.69763184,  1.3018799 ,  2.1972656 ,  3.2025146 ,  4.2022705 ,
         5.2020264 ,  6.201782  ,  7.201538  ,  8.201294  ,  9.20105   ,
        10.997314  , 12.502441  , 14.002075  , 16.001587  , 18.500977  ,
        20.00061   ]
sitecoords = (-87.34821897, 34.65507505, 178.0) 
beamwidth = 1.0

os.environ["WRADLIB_DATA"] = "/any/folder/of/choice"
os.environ["WRADLIB_EARTHDATA_USER"] = "earthdatausrname"
os.environ["WRADLIB_EARTHDATA_PASS"] = "earthdatapass"

has_data = os.environ.get("WRADLIB_EARTHDATA_USER", False)

if has_data:
    ax = wrl.vis.plot_scan_strategy(ranges, elevs, sitecoords, units='km', terrain=True, az=165)
```
#### Expected Output
Matplotlib image containing radar scan strategy plot with terrain height plotted

#### Actual Output

```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-60-6ea448ce3ae3> in <module>
      1 if has_data:
----> 2     ax = wrl.vis.plot_scan_strategy(ranges, elevs, sitecoords, units='km', terrain=True, az=165)

~/anaconda3/envs/vortexse_analysis/lib/python3.8/site-packages/wradlib/vis.py in plot_scan_strategy(ranges, elevs, sitecoords, beamwidth, vert_res, maxalt, range_res, maxrange, units, terrain, az, cg, ax, cmap)
   1068         ll = georef.reproject(xyz, projection_source=rad)
   1069         # (down-)load srtm data
-> 1070         ds = io.get_srtm(
   1071             [ll[..., 0].min(), ll[..., 0].max(), ll[..., 1].min(), ll[..., 1].max()],
   1072         )

~/anaconda3/envs/vortexse_analysis/lib/python3.8/site-packages/wradlib/io/dem.py in get_srtm(extent, resolution, merge)
    153     if not merge:
    154         return demlist
--> 155     dem = gdal.Warp("", demlist, format="MEM")
    156 
    157     return dem

~/anaconda3/envs/vortexse_analysis/lib/python3.8/site-packages/osgeo/gdal.py in Warp(destNameOrDestDS, srcDSOrSrcDSTab, **kwargs)
    675 
    676     if _is_str_or_unicode(destNameOrDestDS):
--> 677         return wrapper_GDALWarpDestName(destNameOrDestDS, srcDSTab, opts, callback, callback_data)
    678     else:
    679         return wrapper_GDALWarpDestDS(destNameOrDestDS, srcDSTab, opts, callback, callback_data)

~/anaconda3/envs/vortexse_analysis/lib/python3.8/site-packages/osgeo/gdal.py in wrapper_GDALWarpDestName(*args)
   4317 def wrapper_GDALWarpDestName(*args):
   4318     """wrapper_GDALWarpDestName(char const * dest, int object_list_count, GDALWarpAppOptions warpAppOptions, GDALProgressFunc callback=0, void * callback_data=None) -> Dataset"""
-> 4319     return _gdal.wrapper_GDALWarpDestName(*args)
   4320 class GDALVectorTranslateOptions(_object):
   4321     """Proxy of C++ GDALVectorTranslateOptions class."""

RuntimeError: No usable source images.
```

#### Problem Description
I figured out the problem has to do with the way the `extent` variable is populated in the `io.get_srtm` function. As of now, the `np.floor` function works fine for longitude values in eastern hemisphere (E) while negative values of longitude in western (W) hemisphere suffer because `np.floor` can lead to `lonmin` and 'latmin` to be the same value. For example, in the case of my example site coordinates, `lonmin = lonmax = -88`. This leads to an empty 'filelist` down the function fetching no suitable SRTM source images to be downloaded.

Instead of the current version of `extent = [int(np.floor(x)) for x in extent]`, a more accurate representation would be `extent = [int(np.floor(abs(x)) * np.sign(x)) for x in extent]` yielding correct limits for lat/lon.

I have tested this for my case and the output looks reasonably good. Submitting a PR alongside this issue.

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

wrl.__version__ = '1.9.1'

</details>

gewitterblitz, 2021-03-03:
I looked at this again this morning and I believe I was wrong in my assessment earlier. The problem was not due to `np.floor` function rather it was the azimuth being provided. I realized this when I could not reproduce the error. This means that the error I saw initially may have occurred when I was providing an azimuth value such that lonmin and lonmax were too close that np.floor gave the same value. 

So yeah, the SRTM downloads will fail only when the azimuth and range cannot resolve at least 1 degree of longitudinal difference between the start and end of radar beam coordinates.

closed 2021-03-03 by gewitterblitz

kmuehlbauer, 2021-03-03:
@gewitterblitz No worries! Thanks for taking the time documenting this.

kmuehlbauer, 2021-03-03:
@gewitterblitz FYI: With wradlib 1.9.1 a earthdata login is needed to download srtm-data in connection with providing these credentials via environment variables.

The anonymous download is not available anymore. But you might have found out already.

gewitterblitz, 2021-03-03:
@kmuehlbauer Yes, I did. Just set up my credentials using:

```
os.environ["WRADLIB_DATA"] = "/directory/of/choice/"
os.environ["WRADLIB_EARTHDATA_USER"] = "earth data username"
os.environ["WRADLIB_EARTHDATA_PASS"] = "earth data password"
```

kmuehlbauer, 2021-03-03:
Great!