wradlib/wradlib #489: wradlib installation problem

babuvanam, 2021-04-24:
i am new to python ,
i have installed wradlib in my system through anaconda prompt ,but when i import it in my program i am getting error:"No mudule named 'wradlib' ",please help me,thanks in advance

babuvanam, 2021-04-24:
here are the screenshots
[image]
[image]
[image]

kmuehlbauer, 2021-04-25:
Hi @babuvanam, how did you call your program?

kmuehlbauer, 2021-04-26:
@babuvanam Can you show the out put of:

`(wradlib) C:\Users\LENOVO>python -c "import wradlib; wradlib.show_versions()"`

babuvanam, 2021-04-26:
Hello @kmuehlbauer,thanks for your response,i executed my program in spyder IDE the following is the error i am getting.
[image]
and output of   (wradlib) C:\Users\LENOVO>python -c "import wradlib; wradlib.show_versions()" is
[image]

kmuehlbauer, 2021-04-26:
@babuvanam That's a common issue with IDE's like `spyder`. You need to make sure, that `spyder` can use  the `wradlib`-environment.

Please read this: https://github.com/spyder-ide/spyder/wiki/Working-with-packages-and-environments-in-Spyder
I would suggest to follow the modular approach: https://github.com/spyder-ide/spyder/wiki/Working-with-packages-and-environments-in-Spyder#the-modular-approach

babuvanam, 2021-04-26:
> @babuvanam That's a common issue with IDE's like `spyder`. You need to make sure, that `spyder` can use the `wradlib`-environment.
>
> Please read this: [...]

now i am getting this error 
[image]
what is environment?how to set it?
my anaconda path is C:\ProgramData\Anaconda3
spyder path is C:\Users\LENOVO
and wradlib path is C:\Users\LENOVO\.conda\envs
how to set these paths?please clarify

kmuehlbauer, 2021-04-26:
@babuvanam You would need to install spyder-kernels as suggested by the error-message.

`(wradlib) C:\Users\LENOVO>conda install spyder-kernels`

I do not use `spyder` and I'm not familiar with this IDE. So you would need to find out yourself. A good starting point is usually the documentation (https://docs.spyder-ide.org/current/index.html). I'm closing this, as this is no wradlib question but a spyder-ide question.

closed 2021-04-26 by kmuehlbauer

babuvanam, 2021-04-26:
> @babuvanam You would need to install spyder-kernels as suggested by the error-message.
>
> `(wradlib) C:\Users\LENOVO>conda install spyder-kernels`
>
> I do not use `spyder` and I'm not familiar with this [...]

ok,thank you