wradlib/wradlib #498: Need to improve Max-Z (CAPPI) plots, Also if you could make it compatible with cartopy as well as raster files (DEM, SHP)

syedhamidali, 2021-07-14:
Example;
[image]

kmuehlbauer, 2021-08-05:
@syedhamidali Thanks for opening this issue. 

We would need to completely free wradlib from any oldstyle `pylab` style  plotting. So if anyone wants to take the lead here...

syedhamidali, 2021-08-05:
I'm in intersted!!! Would love to do that .. so that my programming skills will also improve

syedhamidali, 2021-08-05:
I have improved it according to my need, I would like share code and result, it looks good.
[image]

kmuehlbauer, 2021-08-05:
@syedhamidali Yes, please go ahead. This will be a great improvement in wradlib compared to the current solution.

syedhamidali, 2021-08-05:
#!/usr/bin/env python
# coding: utf-8

# Author: Syed Hamid Ali
# windcirculation@gmail.com
# https://github.com/syedhamidali
print('Importing Libraries')
import datetime as dt
tstart = dt.datetime.now()
import numpy as np
import matplotlib.pyplot as plt
import xarray as xr
import wradlib as wrl
import pandas as pd
import geopandas as gpd
import glob
import os
from matplotlib import axes
from matplotlib.ticker import NullFormatter
import cartopy.crs as ccrs
import cartopy.feature as feat
from cartopy.mpl.gridliner import LONGITUDE_FORMATTER, LATITUDE_FORMATTER
from cartopy.io.shapereader import Reader
from cartopy.feature import ShapelyFeature
import warnings
warnings.filterwarnings('ignore')
print('Importing Libraries Done')

# In[5]:

print('Loading Data...')
fdir = os.getcwd()
files = sorted(glob.glob(fdir+'/MDV*PPIVol.nc'))
print('No. of Files: ',len(files))
site = (73.87,18.04,1310)
fname = '/home/PDTC/PDTCHACPL/MCRKalapureddy/Hamid/data/DEM/India_Districts_ADM2_GADM/India_Districts_ADM2_GADM.shp'
shape_feature = ShapelyFeature(Reader(fname).geometries(),
                                ccrs.PlateCarree(), facecolor='none',ec='gray',alpha=0.7)
print('Data Loading Complete')

for filename in files:
    print('Reading File: ', filename)
    vol = wrl.io.open_cfradial1_dataset(filename)
    print('Iterating over sweeps and stacking points')
    swp_list = []
    print('Creating 3D Grid')
    for v in vol:
         # georeference and stack dimensions
        swp = v.pipe(wrl.georef.georeference_dataset).stack(points=["azimuth", "range"])
        swp_list.append(swp)
    # concat sweeps to volume
    vol0 = xr.concat(swp_list, dim="points")
    # Create XYZ Coordinate DataArray
    xyz = xr.concat([vol0.x, vol0.y, vol0.z], dim="xyz").transpose()
    # Create Target 3D Grid
    trgx = np.linspace(xyz[:, 0].min(),xyz[:, 0].max(), 500)
    trgy = np.linspace(xyz[:, 1].min(),xyz[:, 1].max(), 500)
    trgz = np.linspace(0, 20000, 40)
    yy, hh, xx = np.meshgrid(trgy,trgz,trgx)
    trgxyz = np.stack([xx.flatten(), yy.flatten(), hh.flatten()]).T
    # Create Gridder/Interpolator
    trgshape=xx.shape
    gridder = wrl.vpr.CAPPI(polcoords=xyz,
                         gridcoords=trgxyz,
                         gridshape=trgshape,
                         maxrange=125000,
                         minelev=0,
                         maxelev=90,
                         ipclass=wrl.ipol.Nearest)
    print('Interpolating Data into 3D Grids')
[... 64 lines cut ...]
    # plt.grid(axis='y')

    plt.sca(ax_y)
    plt.contourf(trgz/1e3,trgy/1e3,vol_zh.max(axis=2).T,cmap='jet',levels=range(10,60))
    # plt.xlim(0,20)
    ax_y.set_xticks([5,10,15,20])
    # plt.grid(axis='x')

    plt.sca(ax_cnr)
    plt.tick_params(
        axis='both',          # changes apply to the x-axis
        which='both',      # both major and minor ticks are affected
        bottom=False,      # ticks along the bottom edge are off
        top=False,         # ticks along the top edge are off
        left=False,
        right=False,
        labelbottom=False)
    # labels along the bottom edge are off
    plt.text(0.09,0.65,'Max Range: 125 km',fontsize=12)
    plt.text(0.12,0.5,'VPR Height: 20 km',fontsize=12)
    plt.text(0.15,0.3,pd.to_datetime(vol[0].time.values).strftime('%H:%M:%S Z'),weight='bold',fontsize=17)
    plt.text(0.12,0.15,pd.to_datetime(vol[0].time.values).strftime('%d %b, %Y UTC'),fontsize=14)
    ax_xy.set_aspect('auto')
    plt.savefig('plots/'+'D_Shp_Max-Z_'+str(filename[-29:82])+'.jpg',dpi=200,bbox_inches='tight')
print(dt.datetime.now() - tstart)

kmuehlbauer, 2022-01-24:
While researching #545 this resurfaced. The `plot_plan_and_vert` and `plot_max_plan_and_vert` need to be properly refactored/reimplemented.

kmuehlbauer, 2023-09-07:
Plotting functions have been revisited. Additional plot-items could be added afterwards.

closed 2023-09-07 by kmuehlbauer