wradlib/wradlib #499: gis_export: Incorrect geotransform applied in create_raster_dataset

gogglesguy, 2021-07-23:
The code was adapted from the https://github.com/wradlib/wradlib-notebooks/blob/main/notebooks/fileio/wradlib_gis_export_example.ipynb example.

```python
data, meta = wradlib.io.read_radolan_composite(test_filename)
grid = wradlib.georef.get_radolan_grid(1100, 900)
data, grid = wradlib.georef.set_raster_origin(data, grid, 'upper')
data = np.stack((data, data+100, data+1000))
ds = wradlib.georef.create_raster_dataset(data, grid, projection=radolon_proj)
wradlib.io.write_raster_dataset("geotiff.tif", ds, 'GTiff')
```

According to the docs  `get_radolan_grid`  returns edge aligned pixel coordinates, yet `create_raster_dataset` will assume center aligned pixel coordinates because the grid and data match shapes:

https://github.com/wradlib/wradlib/blob/aa4b4cdded2cb0016d012bdaab6abe73f1cd159f/wradlib/georef/raster.py#L613-L618 
\
In this case we get a geotransform of:

```
[-443.9621669218387, 1.0, 0, -3659.1447242645827, 0, -1.0]
```

This **does not** match the expected output of

```
[-443.4621669218387, 1.0, 0, -3658.6447242645827, 0, -1.0]
```

The latter match the expected grid coordinates in the comments:

https://github.com/wradlib/wradlib/blob/aa4b4cdded2cb0016d012bdaab6abe73f1cd159f/wradlib/georef/rect.py#L169-L181

To me it looks like the example is incorrect. Seems you need a `get_radolan_grid` version that returns a nrows+1,ncols+1 grid for this example to work as that is what is expected by create_raster_dataset:

https://github.com/wradlib/wradlib/blob/aa4b4cdded2cb0016d012bdaab6abe73f1cd159f/wradlib/georef/raster.py#L575-L586

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

INSTALLED VERSIONS
------------------
commit: None
python: 3.9.5 (default, May 24 2021, 12:50:35) 
[GCC 11.1.0]
python-bits: 64
OS: Linux
OS-release: 5.12.12-arch1-1
machine: x86_64
processor: 
byteorder: little
LC_ALL: None
LANG: en_US.UTF-8
LOCALE: ('en_US', 'UTF-8')
libhdf5: 1.12.0
libnetcdf: 4.7.4

xarray: 0.18.2
pandas: 1.3.0
numpy: 1.20.3
scipy: 1.7.0
netCDF4: 1.5.7
pydap: None
h5netcdf: 0.11.0
h5py: 3.2.1
Nio: None
zarr: None
cftime: 1.5.0
nc_time_axis: None
PseudoNetCDF: None
rasterio: 1.2.6
cfgrib: None
iris: None
bottleneck: None
dask: 2021.07.0
distributed: None
matplotlib: 3.4.2
cartopy: None
seaborn: None
numbagg: None
pint: None
setuptools: 57.0.0
pip: 20.3.4
conda: None
pytest: 6.2.4
IPython: 7.24.1
sphinx: None

wradlib: 1.10.3-dev0+g2912126

</details>

kmuehlbauer, 2021-07-25:
@gogglesguy Thanks for catching this. The major problem is that RADOLAN itself is defined by the lower left gridpoints of each cell. Users should be able to just retrieve these point (nrows, ncols). For the creation of raster Datasets we would need all edge points (nrows+1, ncols+1). Users who are interested in the center values would also need a new output (nrows, ncols).

We could add a kwarg `mode` which could take 'radolan' (default) , 'center' or 'edge'.

kmuehlbauer, 2021-08-11:
@gogglesguy There is a proposed fix in #502. Your example code would work with both `mode='center'` or `mode='edge'` in the call to `get_radolan_grid`. I would appreciate if you could give it a try. 

The example needs fixing anyway. This will surface in wradlib 1.11.

PR #502 (2021-08-16): ENH: add `mode`-kwarg to radolan coordinates/grid functions
- fixes #499 (partly)
- add `mode` kwarg to `wradlib.georef.rect.get_radolan_coordinates`
- add `mode` kwarg to `wradlib.georef.rect.get_radolan_grid`

- default is `mode='radolan'` which returns the lower left corner coordinates (nrows, ncols)
- `mode='center'` returns center coordinates (nrows, ncols)
- `mode='edge'` returns edge coordinates (nrows+1, ncols+1)
closed 2021-08-16 by kmuehlbauer