wradlib/wradlib #508: Can't install wradlib after ransomware attack

Miron025, 2021-08-19:
Hi I was infected by .guer ransomware virus. I allready deleted all guer files and reinstalled python/anaconda and all its necessary files, everything works fine but when I tring to install wradlib I still getting this error. Wradlib worked perfectly fine before virus infection. How this can be fixed? Thanks for the help: 

(wradlib) C:\Users\HP>conda install wradlib
Collecting package metadata (current_repodata.json): done
Solving environment: failed with current_repodata.json, will retry with next repodata source.
Initial quick solve with frozen env failed.  Unfreezing env and trying again.
Solving environment: failed with current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): -Exception in thread Thread-4:
Traceback (most recent call last):
  File "C:\Users\HP\Anaconda2\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Users\HP\Anaconda2\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\Users\HP\Anaconda2\lib\site-packages\conda\common\io.py", line 419, in _start_spinning
    self.fh.write('\b' * self._indicator_length)
  File "C:\Users\HP\Anaconda2\lib\site-packages\colorama\ansitowin32.py", line 41, in write
    self.__convertor.write(text)
  File "C:\Users\HP\Anaconda2\lib\site-packages\colorama\ansitowin32.py", line 162, in write
    self.write_and_convert(text)
  File "C:\Users\HP\Anaconda2\lib\site-packages\colorama\ansitowin32.py", line 190, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "C:\Users\HP\Anaconda2\lib\site-packages\colorama\ansitowin32.py", line 195, in write_plain_text
    self.wrapped.write(text[start:end])
  File "C:\Users\HP\Anaconda2\lib\site-packages\colorama\ansitowin32.py", line 41, in write
    self.__convertor.write(text)
  File "C:\Users\HP\Anaconda2\lib\site-packages\colorama\ansitowin32.py", line 162, in write
    self.write_and_convert(text)
  File "C:\Users\HP\Anaconda2\lib\site-packages\colorama\ansitowin32.py", line 190, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "C:\Users\HP\Anaconda2\lib\site-packages\colorama\ansitowin32.py", line 195, in write_plain_text
    self.wrapped.write(text[start:end])
IOError: [Errno 0] Error

kmuehlbauer, 2021-08-19:
@Miron025 What's the output of "python --version" both in the base and wradlib environment. It looks like at least the base env is Python2.

Miron025, 2021-08-20:
> @Miron025 What's the output of "python --version" both in the base and wradlib environment. It looks like at least the base env is Python2.

(base) C:\Users\HP>python --version
Python 2.7.16 :: Anaconda, Inc.

(wradlib) C:\Users\HP>(base) C:\Users\HP>python --version
C:\Users\HP was unexpected at this time.

(wradlib) C:\Users\HP>Python 2.7.16 :: Anaconda, Inc.
Python: can't open file '2.7.16': [Errno 2] No such file or directory

Its almost 4 years when I installed wradlib and I used it automatically and it seems I forget some parts of its installation :D, but after that virus infection it seems that there is still a problem with some features of python even after complete reset...I considering completely reinstalling my OS, but its the last option (a lot of files and data here it will take too much time to make new backups)

Miron025, 2021-08-20:
> @Miron025 What's the output of "python --version" both in the base and wradlib environment. It looks like at least the base env is Python2.

I reinstalled everything and now I have py v3.9.6 but stoill getting the same "colorama" error and stuck on solving environment, I have all dependicies everything is ok but still getting this error I must made a small/big mistake somewhere during the process

kmuehlbauer, 2021-08-20:
@Miron025 Aside from the problems with Anaconda right now, I strongly encourage you to (re) install a fresh OS. You will never know if your cleaning was successful to the last bit.

I'm sorry but this error is out of scope for `wradlib`. I would recommend to ask on the Anaconda issue tracker (https://github.com/ContinuumIO/anaconda-issues). It would be good to search there for this colorama error before asking. Best of luck!

closed 2021-08-20 by kmuehlbauer