wradlib/wradlib #509: ValueError: unrecognized engine odim must be one of: ['netcdf4', 'h5netcdf', 'scipy', 'store']

Guruprasadhegde, 2021-08-19:
Sample code from one of the tutorials from the 'https://docs.wradlib.org/en/stable/notebooks/fileio/wradlib_load_DWD_opendata_volumes.html' throwing 'ValueError: unrecognized engine odim must be one of: ['netcdf4', 'h5netcdf', 'scipy', 'store']' Is there a way to include the odim engine?

#### MCVE Code Sample

```python
# Your code here
for r,v in zip(fpath,fpath_1):
    ds0 = wrl.io.open_odim_mfdataset(r, group="dataset1",
                                     concat_dim="time",
                                     combine="nested",
                                    )
    ds1 = wrl.io.open_odim_mfdataset(v, group="dataset1",
                                     concat_dim="time",
                                     combine="nested",
                                    )

    vol.append(xr.merge([ds0, ds1], combine_attrs="override"))
    vol.sort(key=lambda x: x.time.min().values)

#### Expected Output

#### Problem Description
<!-- this should explain why the current behavior is a problem and why the expected output is a better solution -->

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

<!-- Paste the output of wrl.show_versions() here -->

</details>

kmuehlbauer, 2021-08-19:
@Guruprasadhegde Please provide the output of 

```python
>>> import wradlib as wrl
>>> wrl.show_versions()
```

This was first introduced in wradlib version `1.10.0`.

Guruprasadhegde, 2021-08-20:
@kmuehlbauer Please find below the output from wrl.show_versions()
```python
>>> import wradlib as wrl
>>> wrl.show_versions()
```
INSTALLED VERSIONS
------------------
commit: None
python: 3.8.10 | packaged by conda-forge | (default, May 11 2021, 06:25:23) [MSC v.1916 64 bit (AMD64)]
python-bits: 64
OS: Windows
OS-release: 10
machine: AMD64
processor: Intel64 Family 6 Model 165 Stepping 2, GenuineIntel
byteorder: little
LC_ALL: None
LANG: None
LOCALE: ('English_United Kingdom', '1252')
libhdf5: 1.10.6
libnetcdf: 4.6.1

xarray: 0.19.0
pandas: 1.3.2
numpy: 1.21.2
scipy: 1.7.1
netCDF4: 1.5.7
pydap: None
h5netcdf: 0.11.0
h5py: 3.3.0
Nio: None
zarr: None
cftime: 1.5.0
nc_time_axis: None
PseudoNetCDF: None
rasterio: None
cfgrib: None
iris: None
bottleneck: None
dask: None
distributed: None
matplotlib: 3.4.3
cartopy: 0.18.0
seaborn: None
numbagg: None
pint: 0.17
setuptools: 57.4.0
pip: 21.2.4
conda: None
pytest: None
IPython: 7.26.0
sphinx: None

wradlib: 1.10.4-dev0+g64acd7f

kmuehlbauer, 2021-08-20:
@Guruprasadhegde For this to work, you will need to install `dask` (open_odim_mfdataset). There should have been warnings like below, before the traceback.

```
/home/kai/miniconda3/envs/wrl_test/lib/python3.8/site-packages/xarray/backends/plugins.py:61: RuntimeWarning: Engine 'cfradial1' loading failed:
The 'dask' distribution was not found and is required by the application
  warnings.warn(f"Engine {name!r} loading failed:\n{ex}", RuntimeWarning)
/home/kai/miniconda3/envs/wrl_test/lib/python3.8/site-packages/xarray/backends/plugins.py:61: RuntimeWarning: Engine 'cfradial2' loading failed:
The 'dask' distribution was not found and is required by the application
  warnings.warn(f"Engine {name!r} loading failed:\n{ex}", RuntimeWarning)
/home/kai/miniconda3/envs/wrl_test/lib/python3.8/site-packages/xarray/backends/plugins.py:61: RuntimeWarning: Engine 'gamic' loading failed:
The 'dask' distribution was not found and is required by the application
  warnings.warn(f"Engine {name!r} loading failed:\n{ex}", RuntimeWarning)
/home/kai/miniconda3/envs/wrl_test/lib/python3.8/site-packages/xarray/backends/plugins.py:61: RuntimeWarning: Engine 'odim' loading failed:
The 'dask' distribution was not found and is required by the application
  warnings.warn(f"Engine {name!r} loading failed:\n{ex}", RuntimeWarning)
/home/kai/miniconda3/envs/wrl_test/lib/python3.8/site-packages/xarray/backends/plugins.py:61: RuntimeWarning: Engine 'radolan' loading failed:
The 'dask' distribution was not found and is required by the application
  warnings.warn(f"Engine {name!r} loading failed:\n{ex}", RuntimeWarning
```
So this should be fixed by installing dask.

Just one more question. What are those `r` and `v` file objects of yours? You might not need the current construction. Good chance, that one call to `open_odim_mfdataset` with the proper filestring/path-object.

Guruprasadhegde, 2021-08-20:
Thanks a lot, it works now! 
Also your right about r and v file objects, I was able to handle it with a single call. Thanks again.

kmuehlbauer, 2021-08-20:
@Guruprasadhegde You're welcome and welcome to the wradlib community. Great to hear it works now. Do you think that mentioning `dask` in the [open_odim_mfdataset docstring](https://docs.wradlib.org/en/stable/generated/wradlib.io.hdf.open_odim_mfdataset.html) would have helped in the first place?

Guruprasadhegde, 2021-08-20:
Yes I think it would have helped because from the error message it is really difficult to figure out that `dask` is missing.

kmuehlbauer, 2021-08-20:
Would you mind opening another issue with that special request (adding `dask` to all `open_*_mfdataset` functions)? 

Closing this as resolved.

closed 2021-08-20 by kmuehlbauer