wradlib/wradlib #511: Problem with loading ODIM DWD

Miron025, 2021-08-23:
Everything working fine in Wradlib, except this one, I followed instructions here: https://docs.wradlib.org/en/1.10.0/notebooks/fileio/wradlib_load_DWD_opendata_volumes.html

but after reading into xarray powered structure i get an error:

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_720/2473223119.py in <module>
      1 vol = wrl.io.RadarVolume()
      2 for r, v in zip(volume_reflectivity, volume_velocity):
----> 3     ds0 = wrl.io.open_odim_mfdataset(r, group="dataset1",
      4                                      concat_dim="time",
      5                                      combine="nested",

~\anaconda3\envs\wradlib\lib\site-packages\wradlib\io\hdf.py in open_odim_mfdataset(filename_or_obj, group, **kwargs)
    134     raise_on_missing_xarray_backend()
    135     kwargs["group"] = group
--> 136     return open_radar_mfdataset(filename_or_obj, engine="odim", **kwargs)
    137 
    138 

~\anaconda3\envs\wradlib\lib\site-packages\wradlib\io\xarray.py in open_radar_mfdataset(paths, **kwargs)
   1900         pass
   1901 
-> 1902     ds = [
   1903         xr.open_mfdataset(
   1904             patharr.tolist(),

~\anaconda3\envs\wradlib\lib\site-packages\wradlib\io\xarray.py in <listcomp>(.0)
   1901 
   1902     ds = [
-> 1903         xr.open_mfdataset(
   1904             patharr.tolist(),
   1905             engine=engine,

~\anaconda3\envs\wradlib\lib\site-packages\xarray\backends\api.py in open_mfdataset(paths, chunks, concat_dim, compat, preprocess, engine, data_vars, coords, combine, parallel, join, attrs_file, combine_attrs, **kwargs)
    911         getattr_ = getattr
    912 
--> 913     datasets = [open_(p, **open_kwargs) for p in paths]
    914     closers = [getattr_(ds, "_close") for ds in datasets]
    915     if preprocess is not None:

~\anaconda3\envs\wradlib\lib\site-packages\xarray\backends\api.py in <listcomp>(.0)
    911         getattr_ = getattr
    912 
--> 913     datasets = [open_(p, **open_kwargs) for p in paths]
    914     closers = [getattr_(ds, "_close") for ds in datasets]
    915     if preprocess is not None:

~\anaconda3\envs\wradlib\lib\site-packages\xarray\backends\api.py in open_dataset(filename_or_obj, engine, chunks, cache, decode_cf, mask_and_scale, decode_times, decode_timedelta, use_cftime, concat_characters, decode_coords, drop_variables, backend_kwargs, *args, **kwargs)
    481         engine = plugins.guess_engine(filename_or_obj)
    482 
--> 483     backend = plugins.get_backend(engine)
    484 
    485     decoders = _resolve_decoders_kwargs(

~\anaconda3\envs\wradlib\lib\site-packages\xarray\backends\plugins.py in get_backend(engine)
    154         engines = list_engines()
    155         if engine not in engines:
--> 156             raise ValueError(
    157                 f"unrecognized engine {engine} must be one of: {list(engines)}"
    158             )

ValueError: unrecognized engine odim must be one of: ['netcdf4', 'h5netcdf', 'scipy', 'store']

kmuehlbauer, 2021-08-23:
@Miron025 Might be the same

Miron025, 2021-08-23:
> @Miron025 Might be the same problem as issue #509, missing 'dask'.
> 
> For opening data using '*_mfdataset' you need to install 'dask'.

Thanks a lot, it worked, thanks again...And now final problem that occured today with processing volumes from DWD:

radar = 'ESS'
DBZH = 'sweep_vol_z'
VRADH = 'sweep_vol_v'

opendata_url1 = (f"https://opendata.dwd.de/weather/radar/sites/{DBZH}/{radar.lower()}/hdf5/")
with urllib.request.urlopen(opendata_url1) as url_request:
    response = url_request.read().decode("utf-8")

parser.links = []
parser.feed(response)
filelist1 = parser.links[1:]

filelist1.sort(key=lambda x: x.split('-')[2])
filelist1.reverse()

opendata_url2 = (f"https://opendata.dwd.de/weather/radar/sites/{VRADH}/{radar.lower()}/hdf5/")
with urllib.request.urlopen(opendata_url2) as url_request:
    response = url_request.read().decode("utf-8")

parser.links = []
parser.feed(response)
filelist2 = parser.links[1:]

filelist2.sort(key=lambda x: x.split('-')[2])
filelist2.reverse()

it ended with: list index out of range
 not sure whats wrong, thanks for the help
Miron from the Czech Republic

kmuehlbauer, 2021-08-23:
@Miron025 Great you got it working! You are using some outdated example. DWD changed the location of the files. You would now need to change your path's something like this: 

```python
filt = 'filter_polarimetric' #or 'filter_simple'
opendata_url1 = (f"https://opendata.dwd.de/weather/radar/sites/{DBZH}/{radar.lower()}/hdf5/{filt}")
```

Miron025, 2021-08-23:
> @Miron025 Great you got it working! You are using some outdated example. DWD changed the location of the files. You would now need to change your path's something like this:
>
> ```python
> filt = [...]

Great worked, but:  for f in filelist1[:10]:
    urllib.request.urlretrieve(os.path.join(opendata_url1, f), f)

for f in filelist2[:10]:
    urllib.request.urlretrieve(os.path.join(opendata_url2, f), f)

ended with: http error

HTTPError                                 Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_7312/3008908599.py in <module>
      1 for f in filelist1[:10]:
----> 2     urllib.request.urlretrieve(os.path.join(opendata_url1, f), f)
      3 
      4 for f in filelist2[:10]:
      5     urllib.request.urlretrieve(os.path.join(opendata_url2, f), f)

~\anaconda3\envs\wradlib\lib\urllib\request.py in urlretrieve(url, filename, reporthook, data)
    245     url_type, path = _splittype(url)
    246 
--> 247     with contextlib.closing(urlopen(url, data)) as fp:
    248         headers = fp.info()
    249 

~\anaconda3\envs\wradlib\lib\urllib\request.py in urlopen(url, data, timeout, cafile, capath, cadefault, context)
    220     else:
    221         opener = _opener
--> 222     return opener.open(url, data, timeout)
    223 
    224 def install_opener(opener):

~\anaconda3\envs\wradlib\lib\urllib\request.py in open(self, fullurl, data, timeout)
    529         for processor in self.process_response.get(protocol, []):
    530             meth = getattr(processor, meth_name)
--> 531             response = meth(req, response)
    532 
    533         return response

~\anaconda3\envs\wradlib\lib\urllib\request.py in http_response(self, request, response)
    638         # request was successfully received, understood, and accepted.
    639         if not (200 <= code < 300):
--> 640             response = self.parent.error(
    641                 'http', request, response, code, msg, hdrs)
    642 

~\anaconda3\envs\wradlib\lib\urllib\request.py in error(self, proto, *args)
    567         if http_err:
    568             args = (dict, 'default', 'http_error_default') + orig_args
--> 569             return self._call_chain(*args)
    570 
    571 # XXX probably also want an abstract factory that knows when it makes

~\anaconda3\envs\wradlib\lib\urllib\request.py in _call_chain(self, chain, kind, meth_name, *args)
    500         for handler in handlers:
    501             func = getattr(handler, meth_name)
--> 502             result = func(*args)
    503             if result is not None:
    504                 return result

~\anaconda3\envs\wradlib\lib\urllib\request.py in http_error_default(self, req, fp, code, msg, hdrs)
    647 class HTTPDefaultErrorHandler(BaseHandler):
    648     def http_error_default(self, req, fp, code, msg, hdrs):
--> 649         raise HTTPError(req.full_url, code, msg, hdrs, fp)
    650 
    651 class HTTPRedirectHandler(BaseHandler):

HTTPError: HTTP Error 404: Not Found

(I hope this is the last one from me :D Thanks

kmuehlbauer, 2021-08-23:
@Miron025 Did you adapt opendata_url2 too?

kmuehlbauer, 2021-08-23:
@Miron025 It would be great if you could provide a minimal self-contained code snipped, which could be tested. It would be much easier to debug.

Miron025, 2021-08-23:
> @Miron025 It would be great if you could provide a minimal self-contained code snipped, which could be tested. It would be much easier to debug.

Ok here it is as complete code:

import wradlib as wrl
import warnings
warnings.filterwarnings('ignore')
import matplotlib.pyplot as pl
import numpy as np
import xarray as xr
try:
    get_ipython().magic("matplotlib inline")
except:
    pl.ion()
from wradlib.io.xarray import CfRadial, OdimH5

# In[2]:

import urllib
import os
import io
import glob

# In[3]:

from html.parser import HTMLParser

class DWDHTMLParser(HTMLParser):
    def handle_starttag(self, tag, attrs):
        if tag != 'a':
            return
        self.links.append(attrs[0][1])

parser = DWDHTMLParser()

# In[5]:

radar = 'DRS'
DBZH = 'sweep_vol_z'
VRADH = 'sweep_vol_v'

filt = 'filter_polarimetric' 
opendata_url1 = (f"https://opendata.dwd.de/weather/radar/sites/{DBZH}/{radar.lower()}/hdf5/{filt}")
with urllib.request.urlopen(opendata_url1) as url_request:
    response = url_request.read().decode("utf-8")

parser.links = []
parser.feed(response)
filelist1 = parser.links[1:]

filelist1.sort(key=lambda x: x.split('-')[2])
filelist1.reverse()

filt = 'filter_polarimetric' 
opendata_url2 = (f"https://opendata.dwd.de/weather/radar/sites/{VRADH}/{radar.lower()}/hdf5/{filt}")
with urllib.request.urlopen(opendata_url2) as url_request:
    response = url_request.read().decode("utf-8")

parser.links = []
parser.feed(response)
filelist2 = parser.links[1:]

filelist2.sort(key=lambda x: x.split('-')[2])
filelist2.reverse()

# In[6]:

for f in filelist1[:10]:
    print(f)

# In[7]:

for f in filelist2[:10]:
    print(f)

# In[8]:

flist = glob.glob('ras07*')
for f in flist:
    os.remove(f)

# In[9]:

for f in filelist1[:10]:
    urllib.request.urlretrieve(os.path.join(opendata_url1, f), f)

for f in filelist2[:10]:
    urllib.request.urlretrieve(os.path.join(opendata_url2, f), f)

kmuehlbauer, 2021-08-24:
@Miron025 Your code works for me. Could you please share the output of `wradlib.show_versions()`.

Miron025, 2021-08-24:
> @Miron025 Your code works for me. Could you please share the output of `wradlib.show_versions()`.

INSTALLED VERSIONS
------------------
commit: None
python: 3.8.10 | packaged by conda-forge | (default, May 11 2021, 06:25:23) [MSC v.1916 64 bit (AMD64)]
python-bits: 64
OS: Windows
OS-release: 10
machine: AMD64
processor: AMD64 Family 21 Model 112 Stepping 0, AuthenticAMD
byteorder: little
LC_ALL: None
LANG: None
LOCALE: ('Czech_Czechia', '1250')
libhdf5: 1.10.6
libnetcdf: 4.7.4

xarray: 0.19.0
pandas: 1.3.2
numpy: 1.21.2
scipy: 1.7.1
netCDF4: 1.5.6
pydap: None
h5netcdf: 0.11.0
h5py: 3.3.0
Nio: None
zarr: None
cftime: 1.5.0
nc_time_axis: None
PseudoNetCDF: None
rasterio: None
cfgrib: None
iris: None
bottleneck: None
dask: 2021.08.1
distributed: 2021.08.1
matplotlib: 3.4.3
cartopy: 0.19.0.post1
seaborn: None
numbagg: None
pint: 0.17
setuptools: 57.4.0
pip: 21.2.4
conda: None
pytest: None
IPython: 7.26.0
sphinx: None

wradlib: 1.10.4-dev0+g64acd7f
>>>

kmuehlbauer, 2021-08-24:
@Miron025 Thanks. This looks OK to me so far. Could you also add the output of `conda list`. Either there is a problem with the environment or a problem with the urls.

What do cells 6 and 7 print? 

Could you change cell 9 to:

```python
for f in filelist1[:10]:
    fx = os.path.join(opendata_url1, f)
    print(fx)
    urllib.request.urlretrieve(fx, f)

for f in filelist2[:10]:
    fx = os.path.join(opendata_url2, f)
    print(fx)
    urllib.request.urlretrieve(fx, f)
```
What does that print? Please add complete output.

kmuehlbauer, 2021-08-24:
@Miron025 Got it, hopefully. As you are on windows, `os.path.join` will use a `\` (backslash) instead of a `/` (slash) to concatenate the url, which does result in the http error (not found):

You would need to do something like:
```python
for f in filelist1[:10]:
    fx = "/".join([opendata_url1, f])
    print(fx)
    urllib.request.urlretrieve(fx, f)

for f in filelist2[:10]:
    fx = "/".join([opendata_url2, f])
    print(fx)
    urllib.request.urlretrieve(fx, f)

```

Miron025, 2021-08-24:
> @Miron025 Got it, hopefully. As you are on windows, `os.path.join` will use a `\` (backslash) instead of a `/` (slash) to concatenate the url, which does result in the http error (not found):
>
> You [...]

Thanks for your assistance, it worked now...unfortunatelly now there is nothing interesting over Germany right now, I must wait for some storms to enjoy the beauty of radar volumes...thanks again.

Miron025, 2021-08-24:
> @Miron025 Got it, hopefully. As you are on windows, `os.path.join` will use a `\` (backslash) instead of a `/` (slash) to concatenate the url, which does result in the http error (not found):
>
> You [...]

Is there any way how to open saved volume files? like open fpath//C/RADAR/namefile etc...I cant find it anywhere. I have a lots of files  saved on my external drive such as some supercells and others in Zvol and Vvol fromat from DWD opendata source.

kmuehlbauer, 2021-08-24:
@Miron025 You are welcome. This should have been caught earlier. I had some gut feeling about the url, but only after I saw you are on windows. This need to be adapted in the examples too.

kmuehlbauer, 2021-08-24:
> Is there any way how to open saved volume files? like open fpath//C/RADAR/namefile etc..
Absolutely.

For single files:
https://docs.wradlib.org/en/stable/generated/wradlib.io.hdf.open_odim_dataset.html

For mulltiple files:
https://docs.wradlib.org/en/stable/generated/wradlib.io.hdf.open_odim_mfdataset.html

It's the same thing actually, not depending if the files are from the net or local files.

kmuehlbauer, 2021-08-25:
@Miron025 I'll close this issue as resolved. If you run into any other issue, please open a new issue.

closed 2021-08-25 by kmuehlbauer