wradlib/wradlib #512: Issues installing on MacOS

Rankicker795, 2021-08-25:
I am currently having issues installing wradlib 1.10 on MacOS. I am trying to install onto a conda environment with Python=3.7. I keep getting ImportErrors for various libraries, no matter if I install dependencies first, or allow pip/conda/source code to install the dependicies. I have also tried installing wradlib 1.8 and get similar issues.

kmuehlbauer, 2021-08-25:
@Rankicker795 Are you on a ARM Mac? Could you show what you the commands you tried including the outputs?

kmuehlbauer, 2021-08-30:
@Rankicker795 Did you solve the issue in the meantime, or does the problem persist?

Rankicker795, 2021-08-30:
@kmuehlbauer I am not sure how but I was able to install it and was able to import it. Thanks for replying. 

One last question. Is your software compatible with NEXRAD Archive binary files? I don't believe it is but thought I would check. 

Again, thanks for the help.

kmuehlbauer, 2021-08-30:
> Is your software compatible with NEXRAD Archive binary files? I don't believe it is but thought I would check.

This is correct. Currently we have no reader for NEXRAD Archive binary files. If you like to see this feature in wradlib, please open a new issue with a feature request.

closed 2021-08-30 by kmuehlbauer