wradlib/wradlib #524: ppi plot: 0 degree azimuth swapped to 180 degrees

StevenKnoopKNMI, 2021-10-05:
I am using the wradlib library to make PPI and RHI plots from wind lidar data, measured by a Leosphere Windcube 200s. I have noticed an issue in the wradlib.vis.plot_ppi module: radial windspeeds around an .azimuth of 0 degrees are plotted around 180 degrees.

Python code consists of three modules:
(1) open_single_leo_nc: opens the Leosphere netcdf files and returns a panda dataframe
(2) wradlib_ppi_data: extracts the radial wind speed (and QC for mask), elevation and azimuth values from the panda daframe and puts them in a numpy arrays
(3) ppi_quicklook: puts the numpy arrays into the wradlib.vis.plot_ppi module to make the figure

```python

import numpy as np
import matplotlib.pyplot as pl
import wradlib
import xarray
import pandas as pd
import warnings
warnings.filterwarnings('ignore')
try:
    get_ipython().magic("matplotlib inline")
except:
    pl.ion()

def open_single_leo_nc(filename):
    # opens netcdf file and returns pandas data frame
    # input is file name as string

    # use x array to open and read the netcdf file
    xds = xarray.open_dataset(filename, decode_times=True).transpose()
    xgrname = xds['sweep_group_name'].values[0]
    xds.close()
    xgr = xarray.open_dataset(filename, group=xgrname, decode_times=False).transpose()
    # convert to pandas data frame
    dfnc = xgr.to_dataframe().reset_index()
    xgr.close()
    dfnc['time'] = pd.to_datetime(dfnc.time,unit='s')
    # set and sort index
    dfnc = dfnc.set_index(['time','range']).sort_index()
    
    # add scan ID information from file name to data frame
    scanID = os.path.split(filename)[1].split('_')[4]
    dfnc['scan_ID'] = int(scanID)
    # remove columns of string (object)
    dfnc = dfnc[dfnc.T[dfnc.dtypes!=object].index]
    
    return dfnc

def wradlib_ppi_data(filename):

    df=open_single_leo_nc(filename)
    df=df.reset_index(level=['range'])
    
    v=np.array(df['radial_wind_speed'])
    status=np.array(df['radial_wind_speed_status'])
    r=np.array(df['range'].unique())
    
    elev=df['elevation'][df['range']==200].values
    az=df['azimuth'][df['range']==200].values
    
    # print(df[['radial_wind_speed','azimuth','elevation']][df['range']==200].head(20))
    elev=round(elev.mean(),1)
    
    vm = np.reshape(
                   v,     # the array to be reshaped
                   (len(az),len(r))  # dimensions of the new array
                  )
    mask = np.reshape(
                   status,     # the array to be reshaped
                   (len(az),len(r))  # dimensions of the new array
                  )
    
    vmx = np.ma.masked_array(vm, mask=np.logical_not(mask))
    
    return vmx,r,az,elev

def ppi_quicklook(filename):
    vmx, r, az, elev = wradlib_ppi_data(filename)
    site = (0, 0, 0)
    fig = pl.figure(figsize=(8, 6))
    maxr=round(max(r)/1000)
    ax,pm=wradlib.vis.plot_ppi(vmx, r/1000, az, elev, site=site, proj='cg', fig=fig, ax=111, func='pcolormesh', rf=1,cmap=pl.cm.gist_rainbow,vmin=vmin,vmax=vmax)
    caax = ax.parasites[0]
    paax = ax.parasites[1]
    
    cb = pl.colorbar(pm, pad=0.075, ax=paax)
    cb.set_label(label='radial wind speed (m/s)',fontsize=16)
    xlim = ax.set_xlim(-maxr, maxr)
    ylim = ax.set_ylim(-maxr, maxr)
    pl.title(filename.split('\\')[-1],y=1.06,fontsize=14)
    caax.set_xlabel('distance (km)',fontsize=18)
    caax.set_ylabel('height (km)',fontsize=18)    
    

    fig.tight_layout()
    pl.show()

ppi_quicklook(filename)
```

#### Expected Output

input is a netcdf file from the Windcube 200S (here attached in the zipped file):
[sample netcdf windcube 200s file.zip](https://github.com/wradlib/wradlib/files/7284849/sample.netcdf.windcube.200s.file.zip)

result is this figure:
[image]

#### Problem Description

#### Version

<details><summary>INSTALLED VERSIONS
------------------
commit: None
python: 3.8.10 | packaged by conda-forge | (default, May 11 2021, 06:25:23) [MSC v.1916 64 bit (AMD64)]
python-bits: 64
OS: Windows
OS-release: 10
machine: AMD64
processor: Intel64 Family 6 Model 158 Stepping 11, GenuineIntel
byteorder: little
LC_ALL: None
LANG: en
LOCALE: ('English_United Kingdom', '1252')
libhdf5: 1.10.6
libnetcdf: 4.7.4

xarray: 0.18.2
pandas: 1.2.5
numpy: 1.21.0
scipy: 1.6.3
netCDF4: 1.5.6
pydap: None
h5netcdf: 0.11.0
h5py: 3.3.0
Nio: None
zarr: None
cftime: 1.5.0
nc_time_axis: None
PseudoNetCDF: None
rasterio: None
cfgrib: None
iris: None
bottleneck: None
dask: 2021.06.2
distributed: 2021.06.2
matplotlib: 3.4.2
cartopy: 0.19.0.post1
seaborn: None
numbagg: None
pint: None
setuptools: 49.6.0.post20210108
pip: 21.1.2
conda: None
pytest: None
IPython: 7.24.1
sphinx: 4.0.2

wradlib: 1.10.1-dev0+g1526903</summary>

</details>

kmuehlbauer, 2021-10-05:
@StevenKnoopKNMI 

Thanks for bringing this up. This is surfacing every once in a while. The main problem is the assumption of the plotting routine, that the azimuth angles are given in increasing order (0-360). Unfortunately that this is not mentioned in the docstring.

If the azimuth are given in increasing order the plotting routine makes sure to concat the pcolormesh at the zero degree line.

You can check this by adding one line:

```python
    # set and sort index
    dfnc = dfnc.set_index(['time','range']).sort_index()
    dfnc = dfnc.sort_values(by='azimuth')  # add this line to sort azimuth
```

Please leave the issue open until I've created issues for all Todo points:

- [ ] update docstring mentioning increasing order of azimuth (PPI) and elevation (RHI)
- [ ] order azimuth in any case within the plotting routine
- [ ] fix cfradial2 reader to allow for `decode_times=False`

Are you aware that you can also read your data directly into a CfRadial2-like xarray Dataset using wradlib? Unfortunately the non-standard `time_reference` handling and some bugs prevents us direct usage.

```python
    ds = wrl.io.open_cfradial2_dataset(filename)
```

StevenKnoopKNMI, 2021-10-05:
thanks. yes, azimuthal angle is running from 170 degrees to either 90 degree or 170 degrees, not 0-360 degrees

kmuehlbauer, 2021-10-05:
It looks like the azimuth covers the whole 360deg. but starts at around 170deg.

[image]

StevenKnoopKNMI, 2021-10-05:
Yes, that is correct, the starting azimuth is 170 deg, but it is a full PPI

[grafik]<https://user-images.githubusercontent.com/5821660/136031263-2f00c8f2-69d9-4380-b1b7-030103e51770.png>

kmuehlbauer, 2021-10-05:
With the added line your code works for me and the image is as it should be. Does it work for you too?

StevenKnoopKNMI, 2021-10-05:
I can try in one hour from now. I will keep you updated.

Steven

StevenKnoopKNMI, 2021-10-05:
I have added this single line to sort the azimuth. lndeed, the feature swap 0-180 degrees is gone, but other features are popping in, I have attached a comparison with or without the sorting. 

[image]
[image]

kmuehlbauer, 2021-10-05:
@StevenKnoopKNMI That looks really strange. I did get a very nice looking image. I'll double check the next day.

kmuehlbauer, 2021-10-06:
@StevenKnoopKNMI My pandas-fu is very limited, so resorting to xarray. I've changed the reading function to:

```python
def open_single_leo_nc(filename):
    # opens netcdf file and returns pandas data frame
    # input is file name as string

    # use x array to open and read the netcdf file
    xds = xarray.open_dataset(filename, decode_times=True).transpose()
    xgrname = xds['sweep_group_name'].values[0]
    xds.close()
    xgr = xarray.open_dataset(filename, group=xgrname, decode_times=False)
    xgr = xgr.sortby("azimuth")  # <-- sort here in xarray
    # convert to pandas data frame
    dfnc = xgr.to_dataframe().reset_index()
    xgr.close()
    dfnc['time'] = pd.to_datetime(dfnc.time,unit='s')
    # set and sort index
    dfnc = dfnc.set_index(['time','range'])#.sort_index() <- skip sorting here
    
    # add scan ID information from file name to data frame
    scanID = os.path.split(filename)[1].split('_')[4]
    dfnc['scan_ID'] = int(scanID)
    # remove columns of string (object)
    dfnc = dfnc[dfnc.T[dfnc.dtypes!=object].index]
    
    return dfnc
```

and get this image:

[image]

StevenKnoopKNMI, 2021-10-06:
but when 

dfnc = dfnc.set_index(['time','range']).sort_index() #<- skip sorting here 

instead of 

dfnc = dfnc.set_index(['time','range'])#.sort_index() <- skip sorting

[image]

kmuehlbauer, 2021-10-06:
@StevenKnoopKNMI 

I'm not sure where the problem is. I've created a minimal example only using xarray/wradlib:

```python
xds = xr.open_dataset(filename, decode_times=True).transpose()
xgrname = xds['sweep_group_name'].values[0]
xds = xds[["latitude", "longitude", "altitude"]]
xgr = xr.open_dataset(filename, group=xgrname, decode_times=False)
ds = xr.merge([xds, xgr]) 
ds = ds.assign_coords(dict(azimuth=ds.azimuth,
                           latitude=ds.latitude,
                           longitude=ds.longitude,
                           altitude=0,
                           elevation=ds.elevation,
                           sweep_mode="azimuth_surveillance")).swap_dims(dict(time="azimuth"))
ds = ds.sortby("azimuth")
ds = ds.pipe(wrl.georef.georeference_dataset)
ds = ds.where(ds.radial_wind_speed_status == True)
ds.radial_wind_speed.plot(x="x", y="y")
```
[image]

kmuehlbauer, 2021-10-06:
> your code gives the same plot whether or not the line ds = ds.sortby("azimuth") is present. but I guess the idea is now to use the final ds to get the right input for wradlib.vis.plot_ppi

Yes, since the plot is created via "x" and "y" coordinates, it's not necessary to sort. But for extraction it is. 

But you can just use wradlib xarray accessor plotting machinery to create a similar plot (like plot_ppi). `plot_ppi` does the same things under the hood. No need to extract anything as numpy arrays. 
```
fig = pl.figure(figsize=(8, 6))
maxr = ds.x.max()
vmin = -20
vmax = 20
display(ds)
pm = ds.radial_wind_speed.wradlib.plot_ppi(proj='cg', fig=fig, ax=111, func='pcolormesh', cmap=pl.cm.gist_rainbow,vmin=vmin,vmax=vmax)
ax = pl.gca()
caax = ax.parasites[0]
paax = ax.parasites[1]

cb = pl.colorbar(pm, pad=0.075, ax=paax)
cb.set_label(label='radial wind speed (m/s)',fontsize=16)
xlim = ax.set_xlim(-maxr, maxr)
ylim = ax.set_ylim(-maxr, maxr)
pl.title(filename.split('/')[-1],y=1.06,fontsize=14)
caax.set_xlabel('distance (m)',fontsize=18)
caax.set_ylabel('height (m)',fontsize=18)    

fig.tight_layout()
pl.show()
```
[image]

kmuehlbauer, 2021-10-06:
Great! You might close the issue, if there are no further problems.

StevenKnoopKNMI, 2021-10-06:
well ... I do see one issue. The example I have used is a full PPI, but I also have PPI scans that don't cover 360 degrees, but for instance 170 degrees to 90 degrees. With the current code, I see that some interpolation is applied in the "non-scanned sector" (which was not the case in my original code), which unwanted. Here the idea might be to simply add a ds with the non-scanned sector (with status values False).

[image]

StevenKnoopKNMI, 2021-10-06:
Yes, adding those two lines worked. Thanks! I will close this issue.

closed 2021-10-06 by StevenKnoopKNMI

kmuehlbauer, 2021-10-06:
> Yes, adding those two lines worked. Thanks! I will close this issue.

Glad it works for you! Thanks for working through this.