wradlib/wradlib #541: Many log entries saying "/opt/conda/envs/wradlib/lib/python3.9/site-packages/wradlib/georef/projection.py:304: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.   if LooseVersion(gdal.VersionInfo("RELEASE_NAME")) >= LooseVersion("3"):"

NicolasTsc, 2022-01-06:
Hello I have a small python backend which converts DWD Radolan data into map tiles.
As a first step I use wradlib to convert my data.
The python backend is deployed using Docker and conda-forge.
I just checked the docker logs and realized the the log output is spammed with the following message over and over again:

[image]

```python
# Your code here
    f = wrl.util.get_wradlib_data_file(path + '/' + radarname)
    data_raw, meta = wrl.io.read_radolan_composite(f)
    proj_osr = wrl.georef.create_osr("dwd-radolan")
    xy_raw = wrl.georef.get_radolan_grid(grid_x, grid_y)
    data, xy = wrl.georef.set_raster_origin(data_raw, xy_raw, 'upper')
    ds = wrl.georef.create_raster_dataset(data, xy, projection=proj_osr)
    if os.path.exists(path + '/' + tifname):
        os.remove(path + '/' + tifname)
    wrl.io.write_raster_dataset(path + '/' + tifname, ds, 'GTiff')
```

#### Expected Output
The log entries of the docker containing not containing the same message over and over again.

#### Problem Description
For obvious reason the logs are hard to read when the same warning is getting printed over and over again.

#### Version

Not really sure how to check but I use an environment.yml and conda / conda-forge to install wradlib to my docker container so I guess it is the most current version.

kmuehlbauer, 2022-01-06:
This is fixed in #539. As a workaround you might silence the `DeprecationWarnings` with:

```python
import warnings
warnings.filterwarnings("ignore", category=DeprecationWarning) 
```

kmuehlbauer, 2022-03-18:
wradlib 1.14.0 is already in the release process and should be available later this day.

closed 2022-03-18 by kmuehlbauer