wradlib/wradlib #549: Data size when importing Rainbow SRI flags (off by 4 bits)

binomaiheu, 2022-02-14:
Hi there ! 

I'm trying to import a Rainbow sri file (composite from 2 radars which I recieved) into wradlib.

I've uploaded the file here : [2014032614250000dBR.zip](https://github.com/wradlib/wradlib/files/8060004/2014032614250000dBR.zip) 

Below is my code & error I encoutered. 

#### MCVE Code Sample
```python
import wradlib as wr

from wradlib.io import read_rainbow

rb = read_rainbow('2014032614250000dBR.sri', loaddata=True)
```

#### Output

```
C:\Tools\Miniconda3\envs\radar\lib\site-packages\wradlib\io\rainbow.py in get_rb_blob_from_string(datastring, blobdict)
    291 
    292     # reshape data
--> 293     data.shape = get_rb_data_shape(blobdict)
    294 
    295     return data

ValueError: cannot reshape array of size 3411904 into shape (1275,1338,2)
```

#### Problem Description

I'm not quite familiar with the rainbow format, but I did do some initial investigations. It seems the import of the data itself works fine, however the problem is when importing the flags. Here wradlib expects the output shape to be (1275, 1338 , 2) (returned by `get_rb_data_shape(blob)` in `get_rb_blob_from_string`), however the `data = map_rb_data(data, datadepth)` returns a 1D array of length 3411904, which is 4 too much ( as 1275*1338*2 = 3411900).  Which (given the flags are unpacked to bits, right ? ) is half a byte ?? 

At first I thought it could have something to do with the 4 bytes which are neglected before decompressing ( as my dataset has  
`compression="qt"``  (see rainbow.py around lines 185 - 188), however this removes 4 bytes before decompressing, so that can't really be it. 

I suppose it must be related to `np.unpackbits(data)` in `map_rb_data(data, datadepth)`, but as I'm unfamiliar with the Rainbow format & especially how the flags are encoded in the sri files, I'm at a bit of a loss here, so I'd appreciate a second (or third for that matter :) pair of eyes what's going wrong here. It could of course be entirely me being stupid. 

Thanks for the extremely usefull package btw !
Bino

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

INSTALLED VERSIONS
------------------
commit: None
python: 3.8.12 | packaged by conda-forge | (default, Oct 12 2021, 21:22:46) [MSC v.1916 64 bit (AMD64)]
python-bits: 64
OS: Windows
OS-release: 10
machine: AMD64
processor: Intel64 Family 6 Model 142 Stepping 12, GenuineIntel
byteorder: little
LC_ALL: None
LANG: None
LOCALE: ('Dutch_Belgium', '1252')
libhdf5: 1.12.1
libnetcdf: 4.8.1

xarray: 0.19.0
pandas: 1.3.4
numpy: 1.21.3
scipy: 1.7.1
netCDF4: 1.5.7
pydap: None
h5netcdf: 0.11.0
h5py: 3.4.0
Nio: None
zarr: None
cftime: 1.5.1
nc_time_axis: None
PseudoNetCDF: None
rasterio: None
cfgrib: None
iris: None
bottleneck: None
dask: 2021.09.1
distributed: 2021.09.1
matplotlib: 3.4.3
cartopy: 0.20.1
seaborn: None
numbagg: None
pint: None
setuptools: 58.2.0
pip: 21.3
conda: None
pytest: None
IPython: 7.28.0
sphinx: None

wradlib: 1.11.1-dev0+ga39d4cd

</details>

kmuehlbauer, 2022-02-14:
Thanks @binomaiheu for reporting this issue. 

I'd say this is a bug. You were on the right track. Obviously it has to do with unpackbits. I'm not really sure whats going on. It looks like we just need to strip of the 4 additional bytes. At least I get nice flagmaps doing this.

binomaiheu, 2022-02-14:
Hi Kai, thanks for the quick reply & having a look already !! Much appreciated.  Will confirm tomorrow. Not sure what the most generic way is to fix it. I.e. whether removing the last 4 bytes of the array is always justified ? But works for me. In any case, I've contacted the people at Leonardo and have asked whether they could send me some specs for the dataformat.

kmuehlbauer, 2022-02-15:
> In any case, I've contacted the people at Leonardo and have asked whether they could send me some specs for the dataformat.

I'll have a look into the documentation too. We should be able to fix this properly.

binomaiheu, 2022-02-15:
Could it be that the "0x0A" is not removed from the blob ? From the manual it would seem the blob data itself is enclosed by a "0x0A" byte, both after the <blob> and before the </blob> and when using compression the 4 bytes which are omitted (also in wradlib) are size information (4 byte integer, big endian). Will have a go at this...

binomaiheu, 2022-02-15:
Ok, I did some further investigation

- The issue is with the flag map, which contains a composite flag which is described in the meta data, in the case of the example, we had 2 flags, of each 1 bit, hence bit depth 2, but this could in theory also be 5 (as in the example given in the manual) as it seems ? 

- I suppose the issue arises since the binary data is first read into a byte array (using `get_rb_blob_data`)  (adhereing to 8 bits) and then converted into a bit array uisng `np.unpackbits`, in the case of the example we expect 1275*1338*2 bits which is 3411900, but the array holds 3411904 (since that is dividable in bytes 3411904/8 = 426488

So if the excess bits are at the end, i suppose it's fine to just chop them off (an error would only affect the flags of one pixel then probably somewhere outside the area of interst anyway ?)

kmuehlbauer, 2022-02-15:
Thanks for that thorough investigation @binomaiheu 

I think the problem is already on the packing side. 

In our case there are 3411900 bits to pack. If they do not fit into the uint8 array, the end is padded with zero bits at the end, see https://numpy.org/doc/stable/reference/generated/numpy.packbits.html#numpy.packbits.

So we just need to make sure to align our sizes (strip off the additional bytes) before setting the shape.

kmuehlbauer, 2022-02-15:
@binomaiheu As this looks like the way to go, are you about to implement your suggestion? PR very welcome.

binomaiheu, 2022-02-15:
ok, will do

closed 2022-02-15 by binomaiheu