wradlib/wradlib #559: Import wradlib - getting a GDAL error

semacil, 2022-02-25:
Hi there,

I am trying to improve my skills in python and regarding the tasks on my new job, I want to work with wradlib.

But eventhough I think that I create an environment and install wradlib, ( via : https://docs.wradlib.org/en/stable/installation.html#anaconda-conda ) I still get an error like this,

>>> import wradlib
Traceback (most recent call last):
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\osgeo\__init__.py", line 29, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 565, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1173, in create_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
ImportError: DLL load failed while importing _gdal: Belirtilen yordam bulunamadı.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\wradlib\__init__.py", line 33, in <module>
    from . import adjust  # noqa
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\wradlib\adjust.py", line 118, in <module>
    from wradlib import ipol, util
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\wradlib\ipol.py", line 51, in <module>
    from wradlib import georef, util, zonalstats
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\wradlib\georef\__init__.py", line 24, in <module>
    from .polar import *
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\wradlib\georef\polar.py", line 31, in <module>
    from wradlib.georef import misc, projection
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\wradlib\georef\projection.py", line 33, in <module>
    from osgeo import gdal, ogr, osr
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\osgeo\__init__.py", line 45, in <module>
    _gdal = swig_import_helper()
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\site-packages\osgeo\__init__.py", line 42, in swig_import_helper
    return importlib.import_module('_gdal')
  File "C:\Users\semacil\Anaconda3\envs\wradlib\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_gdal'
>>>

#### Expected Output

Previously I got an error like, 
" On Windows, with Python >= 3.8, DLLs are no longer imported from the PATH.
If gdalXXX.dll is in the PATH, then set the USE_PATH_FOR_GDAL_PYTHON=YES environment variable
to feed the PATH into os.add_dll_directory(). "

But I solved it with the code " conda env config vars set USE_PATH_FOR_GDAL_PYTHON=YES"
Now, I have an error about GDAL.

Thank you very much for your help.
regards,
Sema.

kmuehlbauer, 2022-02-25:
Hi, please activate the environment from a commandline and provide output of the following command:

$ conda list

Also, please add the steps how you created the environement.

semacil, 2022-02-25:
Hi there, I have already using Anaconda prompt and wradlib environment is also activated,
Here is what I got with $ conda list;

(wradlib) C:\Users\semacil>conda list
# packages in environment at C:\Users\semacil\Anaconda3\envs\wradlib:
#
# Name                    Version                   Build  Channel
blosc                     1.21.0               h0e60522_0    conda-forge
boost-cpp                 1.74.0               h9f4b32c_7    conda-forge
brotli                    1.0.9                h8ffe710_6    conda-forge
brotli-bin                1.0.9                h8ffe710_6    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            h15b3021_1009    conda-forge
cartopy                   0.20.2           py39h689385a_3    conda-forge
certifi                   2021.10.8        py39hcbf5309_1    conda-forge
cfitsio                   4.0.0                hd67004f_0    conda-forge
cftime                    1.5.2            py39h5d4886f_0    conda-forge
curl                      7.81.0               h789b8ee_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
deprecation               2.1.0              pyh9f0ad1d_0    conda-forge
expat                     2.4.6                h39d44d4_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              h1989441_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.29.1           py39hb82d6ee_0    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gdal                      3.4.1            py39hd395a97_4    conda-forge
geos                      3.10.2               h39d44d4_0    conda-forge
geotiff                   1.7.0                h144821f_6    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
h5netcdf                  0.13.1             pyhd8ed1ab_0    conda-forge
h5py                      3.6.0           nompi_py39hd4deaf1_100    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.12.1          nompi_h57737ce_103    conda-forge
icu                       69.1                 h0e60522_0    conda-forge
importlib-metadata        4.11.1           py39hcbf5309_0    conda-forge
importlib_metadata        4.11.1               hd8ed1ab_0    conda-forge
intel-openmp              2022.0.0          h57928b3_3663    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9e                   h8ffe710_0    conda-forge
kealib                    1.4.14               h8995ca9_3    conda-forge
kiwisolver                1.3.2            py39h2e07f2f_1    conda-forge
krb5                      1.19.2               h6da9e4a_3    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      3.0                  h0e60522_0    conda-forge
libblas                   3.9.0              13_win64_mkl    conda-forge
libbrotlicommon           1.0.9                h8ffe710_6    conda-forge
libbrotlidec              1.0.9                h8ffe710_6    conda-forge
libbrotlienc              1.0.9                h8ffe710_6    conda-forge
libcblas                  3.9.0              13_win64_mkl    conda-forge
libcurl                   7.81.0               h789b8ee_0    conda-forge
libdeflate                1.10                 h8ffe710_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libgdal                   3.4.1                h49875e8_4    conda-forge
libglib                   2.70.2               h3be07f2_4    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libkml                    1.3.0             h9859afa_1014    conda-forge
liblapack                 3.9.0              13_win64_mkl    conda-forge
libnetcdf                 4.8.1           nompi_h1cc8e9d_101    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libpq                     14.2                 h1ea2d34_0    conda-forge
librttopo                 1.1.0                hb1df466_9    conda-forge
[... 42 lines cut ...]
semver                    2.13.0             pyh9f0ad1d_0    conda-forge
setuptools                60.9.3           py39hcbf5309_0    conda-forge
shapely                   1.8.0            py39hf063a39_5    conda-forge
six                       1.16.0             pyh6c4a22f_0    conda-forge
sqlite                    3.37.0               h8ffe710_0    conda-forge
tbb                       2021.5.0             h2d74725_0    conda-forge
tiledb                    2.6.2                h47404fa_1    conda-forge
[... cut ...]
xmltodict                 0.12.0                     py_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
zipp                      3.7.0              pyhd8ed1ab_1    conda-forge
zlib                      1.2.11            h8ffe710_1013    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge

semacil, 2022-02-25:
I used these steps while creating the environment,
https://docs.wradlib.org/en/stable/installation.html#anaconda-conda

Add the [conda-forge](https://conda-forge.org/) channel, where ωradlib and its dependencies are located. Read more about the community effort conda-forge:

$ conda config --add channels conda-forge
Use strict channel priority to prevent channel clashes:

$ conda config --set channel_priority strict
Create a new environment from scratch:

$ conda create --name wradlib python=3.8
Activate the ωradlib environment:

$ conda activate wradlib
Install ωradlib and its dependencies:

(wradlib) $ conda install wradlib

kmuehlbauer, 2022-02-25:
Thanks @semacil, that all looks fine. 

How do you invoke the Python session? There might be a chance, that the environment is not correctly activated. Especially GDAL needs a proper activation to get all the environment variables right.

To rule that out, please open a new commandline, activate the wradlib environment, start a python session from the commandline and import wradlib.

kmuehlbauer, 2022-02-25:
Ah, why is the wradlib version so old? Normally the above code should install the most recent version. Try to upgrade wradlib:

$ conda install wradlib==1.13.0

semacil, 2022-02-25:
Thanks for checking. @kmuehlbauer . I am trying to make these changes from PC at my company. I had already made changes on proxy settings at .condarc file but installing procedures always take time.

For example, I use the command tho upgrade, but I am still waiting on this;

(wradlib) C:\Users\semacil>conda install wradlib==1.13.0
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: /

kmuehlbauer, 2022-02-25:
You might install "mamba". That's a fast dropin for conda especially for the solving process.

semacil, 2022-02-25:
Well, I got lots of conflicts for wradlib 1.13.0 . Here are some of them,

"
Package scandir conflicts for:
importlib_metadata -> pathlib2 -> scandir
importlib-metadata -> pathlib2 -> scandir

Package xorg-libxau conflicts for:
pillow -> libxcb[version='>=1.13,<1.14.0a0'] -> xorg-libxau
libxcb -> xorg-libxau

Package lockfile conflicts for:
pip -> lockfile
python=3.9 -> pip -> lockfile

Package munkres conflicts for:
fonttools -> munkres
matplotlib-base -> fonttools[version='>=4.22.0'] -> munkres

Package requests conflicts for:
cartopy -> owslib -> requests[version='>=1.0']
pip -> requests
wradlib==1.13.0 -> requests
python=3.9 -> pip -> requests

Package cftime conflicts for:
netcdf4 -> cftime
wradlib==1.13.0 -> netcdf4 -> cftime
xarray -> netcdf4 -> cftime

Package libxslt conflicts for:
postgresql -> libxslt=1.1
libgdal -> postgresql -> libxslt=1.1

Package libkml conflicts for:
gdal -> libgdal==3.0.2=h1155b67_0 -> libkml[version='>=1.3.0,<1.4.0a0']
libgdal -> libkml[version='>=1.3.0,<1.4.0a0']

Package gdal conflicts for:
wradlib==1.13.0 -> gdal
wradlib==1.13.0 -> cartopy -> gdal[version='>=3.1.2,<3.2.0a0|>=3.1.3,<3.2.0a0']
cartopy -> gdal[version='>=3.1.2,<3.2.0a0|>=3.1.3,<3.2.0a0']

Package postgresql conflicts for:
gdal -> libgdal==3.0.2=ha1b3edf_1 -> postgresql
libgdal -> postgresql

Package cached-property conflicts for:
wradlib==1.13.0 -> h5py -> cached-property
h5netcdf -> h5py -> cached-property
h5py -> cached-property

Package font-ttf-ubuntu conflicts for:
fonts-conda-forge -> font-ttf-ubuntu
fonts-conda-ecosystem -> fonts-anaconda -> font-ttf-ubuntu

Package wheel conflicts for:
pip -> wheel
python=3.9 -> pip -> wheel

Package netcdf4 conflicts for:
wradlib==1.13.0 -> netcdf4
xarray -> netcdf4

Package pandas conflicts for:
xarray -> pandas[version='>=0.18|>=0.18.0|>=0.19.2|>=0.24|>=0.25|>=1.0|>=1.1']
wradlib==1.13.0 -> xarray -> pandas[version='>=0.18|>=0.18.0|>=0.19.2|>=0.24|>=0.25|>=1.0|>=1.1']
xarray -> dask[version='>=0.9'] -> pandas[version='>=0.19.0|>=0.21.0|>=0.23.0|>=0.25.0']

Package xorg-libxdmcp conflicts for:
pillow -> libxcb[version='>=1.13,<1.14.0a0'] -> xorg-libxdmcp
libxcb -> xorg-libxdmcp

Package font-ttf-inconsolata conflicts for:
fonts-conda-forge -> font-ttf-inconsolata
fonts-conda-ecosystem -> fonts-anaconda -> font-ttf-inconsolata

Package pthread-stubs conflicts for:
pillow -> libxcb[version='>=1.13,<1.14.0a0'] -> pthread-stubs
libxcb -> pthread-stubs

Package m2w64-xz conflicts for:
gdal -> libgdal==3.0.2=h3d7c30d_2 -> m2w64-xz
libgdal -> m2w64-xz

Package h5netcdf conflicts for:
wradlib==1.13.0 -> h5netcdf
xarray -> h5netcdf

Package brotli conflicts for:
fonttools -> brotli[version='>=1.0.1']
matplotlib-base -> fonttools[version='>=4.22.0'] -> brotli[version='>=1.0.1']

Package unittest2 conflicts for:
h5netcdf -> h5py -> unittest2
h5py -> unittest2
wradlib==1.13.0 -> h5py -> unittest2

Package kealib conflicts for:
gdal -> libgdal==3.0.2=h3d7c30d_2 -> kealib[version='>=1.4.10,<1.5.0a0|>=1.4.13,<1.5.0a0|>=1.4.14,<1.5.0a0|>=1.4.7,<1.4.8.0a0']
libgdal -> kealib[version='>=1.4.10,<1.5.0a0|>=1.4.13,<1.5.0a0|>=1.4.14,<1.5.0a0|>=1.4.7,<1.4.8.0a0']

Package cached_property conflicts for:
cached-property -> cached_property[version='>=1.5.2,<1.5.3.0a0']
h5py -> cached-property -> cached_property[version='>=1.5.2,<1.5.3.0a0']

Package font-ttf-source-code-pro conflicts for:
fonts-conda-ecosystem -> fonts-anaconda -> font-ttf-source-code-pro
fonts-conda-forge -> font-ttf-source-code-pro

Package fonttools conflicts for:
wradlib==1.13.0 -> matplotlib-base -> fonttools[version='>=4.22.0']
cartopy -> matplotlib-base[version='>=1.5.1'] -> fonttools[version='>=4.22.0']
matplotlib-base -> fonttools[version='>=4.22.0']
Note that strict channel priority may have removed packages required for satisfiability."

I will also try to download mamba. Thanks for the advice.

kmuehlbauer, 2022-02-25:
I've no idea why upgrading wradlib brings up this many conflicts. Also still wondering how you ended up with 1.8.

Maybe just try to create a completely new environment.

semacil, 2022-02-25:
Thanks @kmuehlbauer for the help. I came home and tried to install wradlib via these instructions https://docs.wradlib.org/en/stable/installation.html 
I got errors and I installed some dependencies. Now, I got this error which I have never seen before.
"
(wradlib) C:\Users\semac>python
Python 3.9.10 | packaged by conda-forge | (main, Feb  1 2022, 21:22:07) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import wradlib
C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\wradlib\io\radolan.py:885: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  "add_offset": np.float(0),
C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\wradlib\io\radolan.py:894: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  "add_offset": np.float(0),
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\wradlib\__init__.py", line 33, in <module>
    from . import adjust  # noqa
  File "C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\wradlib\adjust.py", line 118, in <module>
    from wradlib import ipol, util
  File "C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\wradlib\ipol.py", line 51, in <module>
    from wradlib import georef, util, zonalstats
  File "C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\wradlib\zonalstats.py", line 69, in <module>
    from wradlib import georef, io
  File "C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\wradlib\io\__init__.py", line 33, in <module>
    from .xarray import *
  File "C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\wradlib\io\xarray.py", line 103, in <module>
    from xarray.backends.api import _MultiFileCloser, combine_by_coords
ImportError: cannot import name '_MultiFileCloser' from 'xarray.backends.api' (C:\Users\semac\anaconda3\envs\wradlib\lib\site-packages\xarray\backends\api.py)"

I think this time I took right steps to do it but I don't understand why I see these errors.
Thanks for the help,
Kind regards.

semacil, 2022-02-26:
It seems like the problem is solved after updating wradlib to 1.13.0. thank you @kmuehlbauer 
I am using python 3.9 with Anaconda. So I have already created the environment with python 3.9. I am not sure if this is a windows issue but both my personal computer and the computer at my office are windows 10.

I follow the documentation step by step. I don't know why it is needed to update wradlib just after creating the environment.

kmuehlbauer, 2022-02-26:
Glad to hear it works now. Maybe it would be enough to add a sentence to the docs regarding that issue? Do you think this would help users?

closed 2022-02-28 by semacil