wradlib/wradlib #573: VRADH Values??

Miron025, 2022-05-15:
I have a question about right plotting values for DWD VRADH

[image]
[image]

The output is fine...but I can't find anywhere the right values for VRADH, I mean vmin, vmax=?? I tired using m/s, km/h, knots but it seems its still a little bit different in compare to kachelmann public output.

Is there any way how to find the real values that are used for german VRADH? -/+? and then to plot a scalle (Im able to plot a scalle but don't know the values :D ) 

map_proj = ccrs.Mercator(central_longitude=vol.root.longitude.values)
fig = pl.figure(figsize=(100,100))
 
pm = vol[5].isel(time=0).VRADH.pipe(wrl.georef.georeference_dataset).wradlib.plot(ax=gs[i], cmap="NWSVel**",vmin=-20, vmax=20)**

Thanks for the advice :)

MIRON from the Czech Republic

kmuehlbauer, 2022-05-16:
@Miron025 The DWD is available in ODIM_H5 format as packed data. The data is packed into a 8 or 16 bit unsigned integer with connected `gain` and `offset` given in group `what` (see [1]).

For the ODIM xarray backend, these values are transferred into the `.encoding`-property of the DataArray (eg. `VRADH.encoding`, `gain` -> `scale_factor`, `offset` -> `add_offset`). The minimum and maximum velocity would be calculated via this function

quantity_value = offset + gain × raw_value

vmin = -32.13529120559098 + 0.252041499651694 x 0 = -32.13529120559098
vmax = -32.13529120559098 + 0.252041499651694 x 255 = 32.13529120559098

The unit of this result is [m/s] as given in [1].

But note, that these velocities are radial velocities (velocities in direction to the radar). If the actual velocities are lower/higher than vmin/vmax than they will be aliased into that dynamic range. That might explain the differences with Kachelmannwetter. You would need to dealias these velocities. A comprehensive write-up for the current available dealiasing algorithms and implementations is available in UNRAVEL: A Robust Modular Velocity Dealiasing Technique for Doppler Radar [2].

[1] https://www.eumetnet.eu/wp-content/uploads/2019/01/ODIM_H5_v23.pdf p.24 (chapter 4.5)
[2] https://journals.ametsoc.org/view/journals/atot/37/5/jtech-d-19-0020.1.xml

kmuehlbauer, 2022-05-19:
@Miron025 Did you find your way through the information?

Miron025, 2022-05-22:
> [kmuehlbauer](https://github.com/kmuehlbauer) commented [3 days ago](https://github.com/wradlib/wradlib/issues/573#issuecomment-1131596893)
@Miron025 Did you find your way through the information?

[image]

Yes thanks a lot...I used this informations for analysis of German tornado, thanks :)

closed 2022-05-22 by Miron025