wradlib/wradlib #575: DWD VOLUME FILES OPEN

ShelfWall, 2022-05-22:
Greetings!

I recently installed wradlib and everything working fine. But I still have problems with opening these files:

https://opendata.dwd.de/weather/radar/sites/sweep_vol_v/isn/hdf5/filter_polarimetric/

(the VOL) files...Its easy using wetterdients but I'm talking about opening downloaded files from computer.

I tried this: https://docs.wradlib.org/en/1.9.0/notebooks/fileio/wradlib_odim_multi_file_dataset.html

but I get error after:
%%time
vol = wrl.io.xarray.open_odim(f, loader='h5py', chunks={})

ERROR:AttributeError: module 'wradlib.io.xarray' has no attribute 'open_odim'

MY FULL CODE: (untill the error)

# In[14]:

import wradlib as wrl
import warnings
warnings.filterwarnings('ignore')
import matplotlib.pyplot as pl
import numpy as np
import xarray as xr
import os
import glob
try:
    get_ipython().magic("matplotlib inline")
except:
    pl.ion()
    
from wradlib.io.xarray_depr import CfRadial, OdimH5

# In[15]:

import os
import psutil
import gc
process = psutil.Process(os.getpid())

# In[16]:

def memory_usage_psutil():
    # return the memory usage in MB
    rocess = psutil.Process(os.getpid())
    mem = process.memory_full_info().uss / float(1 << 20)
    return mem

# In[17]:

def free_memory():
    mem0 = memory_usage_psutil()
    print(gc.collect())
    proc = psutil.Process()
    mem1 = memory_usage_psutil()
    print("Memory freed: {0:.5f} MB".format((mem0-mem1)))

# In[18]:

def check_open_files(full=False):
    proc = psutil.Process()
    print(len(proc.open_files()))
    if full:
        print(proc.open_files())

# In[19]:

fpath = wrl.util.get_wradlib_data_path()
f = os.path.join(fpath, 'C/SWEEPras07-pcpng01_sweeph5onem_dbzh_00-2022052014253400-fld-10440-hd5')

# In[20]:

mem_start = memory_usage_psutil()
print("Current Memory:", mem_start)

# In[21]:

check_open_files()

# In[26]:

get_ipython().run_cell_magic('time', '', "vol = wrl.io.xarray.open_odim(f, loader='h5py', chunks={})\n")

# In[ ]:

from wradlib.io.xarray_depr import CfRadial, OdimH5

-------------------------------------------
Or can you provide an exemple code how to 
open the VOLUME files from min computer directory. 

Thanks

ShelfWall, 2022-05-22:
I can open the files "VOL" but it showing only one elevation like 25. 
print(vol[0])
<wradlib.XRadTimeSeries>
Dimension(s): (time: 1, azimuth: 360, range: 600)
Elevation(s): (0.8)

but it must have more than one, because I opened these files with wetterdients and it worked...file attached
[VOL H5.zip](https://github.com/wradlib/wradlib/files/8750322/VOL.H5.zip)

kmuehlbauer, 2022-05-23:
@ShelfWall 

First, the `open_odim`-function is deprecated. It should not be used with newer wradlib versions. If you need to use it for some reasons:

https://docs.wradlib.org/en/1.16.0/notebooks/fileio/wradlib_odim_multi_file_dataset.html

Recommended is usage of xarray backends:

https://docs.wradlib.org/en/stable/notebooks/fileio/wradlib_odim_backend.html

Second, DWD radar data is provided in the so called `one sweep one moment` layout. So in one radar file there is only one sweep and one moment. In your above opendata resource there are files like:

- https://opendata.dwd.de/weather/radar/sites/sweep_vol_v/isn/hdf5/filter_polarimetric/ras07-vol5minng01_sweeph5onem_vradh_00-2022052107305700-isn-10873-hd5
- https://opendata.dwd.de/weather/radar/sites/sweep_vol_v/isn/hdf5/filter_polarimetric/ras07-vol5minng01_sweeph5onem_vradh_01-2022052107312000-isn-10873-hd5
- more files for sweeps 02 to 08
- https://opendata.dwd.de/weather/radar/sites/sweep_vol_v/isn/hdf5/filter_polarimetric/ras07-vol5minng01_sweeph5onem_vradh_09-2022052107340200-isn-10873-hd5

Each of these files belongs to the same 5 Minute RadarVolume.

If you want to create a `RadarVolume` you would need to feed all these files to the reading function in a certain manner:

```python
import datetime as dt
import glob
import os
import wradlib as wrl
import xarray as xr

def get_data_path_dwd(inpath):
    """ Get data path (realpep) 
    """
    print(inpath)
    return os.path.join(inpath, '{sweep}/{radar}/{fmt}/{filt}')

def get_file_name_dwd():
    """ Get data path (automount) 
    """
    return ('ras07-{mode}_sweeph5onem_{moment}_{scan}-{year}{month:02d}{day:02d}'
            '{hour:02d}{mintens}[{minones0}-{minones1}]*')

# setup placeholders
inpath = '/home/kai/python/gists/wradlib/shelfwall/opendata.dwd.de/weather/radar/sites/'
sweep = "sweep_vol_v"
mode = "vol5minng01"
mom = "vradh"
radar = "isn"
fmt = "hdf5"
filt = "filter_polarimetric"
date = "2022-05-22"
t = dt.datetime.strptime(date, "%Y-%m-%d")
scan = "*"

# get path
path = os.path.join(get_data_path_dwd(inpath), get_file_name_dwd())
path

# set search path
spath = path.format(sweep=sweep, fmt=fmt, filt=filt,
                    year=t.year, month=t.month, day=t.day, 
                    scan=scan, radar=radar, mode=mode, moment=mom,
                    hour=t.hour, mintens=int(t.minute / 10),
                    minones0=t.minute % 10 + 0,
                    minones1=t.minute % 10 + 4)

# get files
flist = glob.glob(spath)

# create RadarVolume
vol = wrl.io.RadarVolume()

# read files and fill volume
for f in flist:
    vol.append(xr.open_dataset(f, engine="odim", group="dataset1"))
```

ShelfWall, 2022-05-23:
Thanks that was so easy :D thank you again. Ok...is there any way how to view full scans? Does DWD provide full sweeps somewhere? Kachelmann is ok, but these data can't be manipulated it is only a png image.

kmuehlbauer, 2022-05-23:
I'm not sure I get this. What do you mean by "full sweeps"? A bit more information what (and how) you are trying to achieve would be good to get you more help.

ShelfWall, 2022-05-23:
I mean if there is a files where you can plot chosen elevation like VOL=5, 4,3,2,1 (not using wetterdients) but manually downloading the files from DWD for future analysis. Or its not possible and the only way how to look trhough all elevations is to load it online (3 days archive)

kmuehlbauer, 2022-05-24:
No, to my knowledge DWD does not distribute complete volume files per radar. You would need to download the wanted files and create the volume as shown above. You would be able to write this created volume to one file then (`vol.to_odim(outfilename)`). This volume file can be read with the `wrl.io.open_odim_dataset(fname)` directly into a `RadarVolume`.

ShelfWall, 2022-05-25:
Thank you for your answers, I solved this by automatic download (via wetterdients) and direct downloading the chosen radar elevations (auto download 5/min) into the computer and keeping the files for further analysis (or they can be manually deleted if there was nothing special to analyse). These files are exported as single (without map) + gif file (with map) and connected to simple radar viewing app. :)
[image]
 :
[image]

[image]
[image]

closed 2022-05-25 by kmuehlbauer