wradlib/wradlib #582: Wradlib installation issue

akhil24jun, 2022-08-05:
I have installed wradlib using anaconda. However, when i am trying to import the wradlib it is throwonh errors ModuleNotFoundError: No Module named 'requests'
The error message receiwbd along with conda liat and conda info is attached for your reference.
Please suggest.
[image]
[image]
 
[image]
[image]
[image]

kmuehlbauer, 2022-08-05:
@akhil24jun You have a very outdated wradlib version in your environment.

Please run `conda install wradlib=1.16.0` to get the latest version.

kmuehlbauer, 2022-08-05:
Please check the command, it might need double == .

kmuehlbauer, 2022-08-08:
@akhil24jun Do you need further help with that issue?

akhil24jun, 2022-08-08:
Hi, @kmuehlbauer 
I am working to install the wradlib 1.16.0 and will then check whether it's working or not.
I will update regarding the same in sometime,
[image]

kmuehlbauer, 2022-08-08:
@akhil24jun I see that you are installing in the (base) environment. This is not good practise. In fact it is recommended to create a new environment for different use cases.

```
(base) $ conda create -n wradlib python=3.10 wradlib
(base) $ conda activate wradlib
(wradlib) $
```

akhil24jun, 2022-08-08:
@kmuehlbauer Thank you very much for the support.
The latest version is working fine.
[image]

kmuehlbauer, 2022-08-08:
@akhil24jun Great news. If you are running into problems using wradlib, please open issues. I'll close this one.

closed 2022-08-08 by kmuehlbauer