wradlib/wradlib #583: problem with numpy version

SongJaeIn, 2022-08-06:
recently, I installed wradlib via pip.
The installation itself was successful, but following error is occurred when I import this module:
`RuntimeError                              Traceback (most recent call last)
RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xf
`
I am using macOS, and the numpy info is below.

```
Name: numpy
Version: 1.23.1
Summary: NumPy is the fundamental package for array computing with Python.
Home-page: https://www.numpy.org
Author: Travis E. Oliphant et al.
Author-email:
License: BSD
Location: /Users/jaein_song/research/lib/python3.9/site-packages
Requires:
Required-by: basemap, Cartopy, cftime, h5py, matplotlib, netCDF4, pandas, scipy, wradlib, xarray
```

kmuehlbauer, 2022-08-06:
@SongJaeIn I've moved the issue to the wradlib/wradlib repository. 

Two Questions: Which version of wradlib is installed? What Python version are you using?

This error message is displayed if there are packages inside the environment which are compiled using a different numpy version. As wradlib itself is pure Python I would suggest to check the wheels of the dependencies or packages.

molinav, 2022-08-07:
@kmuehlbauer The NumPy API incompatibility issue happened while fixing https://github.com/matplotlib/basemap/issues/547, so it is safe for you to close this issue, since the problem was already solved there.

kmuehlbauer, 2022-08-07:
@Thanks for coming back. Glad it is solved.

kmuehlbauer, 2022-08-07:
Oh the ping went wrong. Thanks for notifying, @molinav .

closed 2022-08-07 by kmuehlbauer