wradlib/wradlib #593: Rading ASCII radolan data files using wradlib

SandeepAllampalli, 2022-10-16:
I am trying to read ASCII format radolan data files using wradlib for my final thesis. However, from the wradlib documentation I understand, it is not possible to read ASCII radolan data files using wradlib as of now. What is the current alternative for reading such files and also looking forward for new enhancements to read ASCII radolan using wradlib.

kmuehlbauer, 2022-10-16:
@SandeepAllampalli Thanks, for this feature request.

This is not a straightforward undertaking, as so much is involved. The major pain point is that the ascii header has only little information:

```
ncols         900
nrows         900
xllcorner     -523462
yllcorner     -4658645
cellsize      1000
NODATA_value  -1
```

We can get the product type and the datetime from the filename, but this prevents us from using in memory buffers. But that might be OK to have to read a file from disk, I think.

I've hacked up a working prototype. I'll see if I can get a PullRequest ready next week.

kmuehlbauer, 2022-10-16:
We can be smart and fake something up here, so we are back in business of on memory objects. Unfortunately we lose lazy loading, but in that case it's probably not needed anyway.

```python
fp = tarfile.open("RW-20221015.tar.gz")
names = fp.getnames()
buf = fp.extractfile(names[0])
wbuf = io.BytesIO(buf.read())
wbuf.name = names[0]
ds = wrl.io.open_radolan_dataset(wbuf)
```

Note: That does not work as of now with ascii data, but with binary data.

kmuehlbauer, 2022-10-16:
@SandeepAllampalli If you want to check my current WIP-branch with added ascii reading you would need to install that PR branch: https://github.com/wradlib/wradlib/pull/594

Let me know how it goes.

SandeepAllampalli, 2022-10-16:
@kmuehlbauer How to install PR branch?

kmuehlbauer, 2022-10-16:
`$ python -m pip install git+https://github.com/kmuehlbauer/wradlib.git@radolan-ascii`

SandeepAllampalli, 2022-10-16:
[image]
I get this error.

kmuehlbauer, 2022-10-16:
Activate your environment first. Do never work in base environment, that can mess up your conda installation.

SandeepAllampalli, 2022-10-16:
I see. I was working in the base environment all this while. I will uninstall the package in the base environment and activate a new environment for wradlib and reinstall it in that environment. Thanks for the info.

PR #594 (2022-10-19): WIP: ENH: read RADOLAN ascii format
ASCII files currently not tested.

closes #591
closes #593
closed 2022-10-19 by kmuehlbauer