wradlib/wradlib #596: wradlib.io.misc.get_radiosonde error when getting data from station without 'Station identifier'

JulianGiles, 2022-10-24:
When using the `wradlib.io.misc.get_radiosonde` function, an error like `"ValueError: invalid literal for int() with base 10: '180922/1200'"` is raised for stations which do not have a 'Station identifier' parameter.

#### MCVE Code Sample

```python
import wradlib as wrl
import datetime as dt

start_time = dt.datetime.strptime("20180923", "%Y%m%d")

rs_id = 10393
rs_data, rs_meta = wrl.io.get_radiosonde(rs_id, start_time)

```

#### Expected Output
A numpy.ndarray of data and a dict of metadata

#### Problem Description

When selecting a station that does not have a 'Station identifier', the following error occurs:
```python
File "/home/mambaforge/envs/wradlib/lib/python3.10/site-packages/wradlib/io/misc.py", line 198, in get_radiosonde
    v = int(v)

ValueError: invalid literal for int() with base 10: '180922/1200'
```

When selecting a station with 'Station identifier', e.g. 10410 'EDZE', the error does not occur. From what I see in the `misc.py` file, the issue is related to how the function handles the list of metadata that is created, when the list is missing the 'Station identifier' all items are shifted and the information is incorrectly processed.

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

INSTALLED VERSIONS
------------------
commit: None
python: 3.10.5 | packaged by conda-forge | (main, Jun 14 2022, 07:04:59) [GCC 10.3.0]
python-bits: 64
OS: Linux
OS-release: 5.14.21-150400.24.21-default
machine: x86_64
processor: x86_64
byteorder: little
LC_ALL: None
LANG: de_DE.UTF-8
LOCALE: ('de_DE', 'UTF-8')
libhdf5: 1.12.2
libnetcdf: 4.8.1

xarray: 2022.3.0
pandas: 1.4.3
numpy: 1.22.4
scipy: 1.8.1
netCDF4: 1.6.1
pydap: None
h5netcdf: 1.0.1
h5py: 3.7.0
Nio: None
zarr: None
cftime: 1.6.1
nc_time_axis: 1.4.1
PseudoNetCDF: None
rasterio: None
cfgrib: None
iris: None
bottleneck: 1.3.5
dask: 2022.7.0
distributed: 2022.7.0
matplotlib: 3.5.2
cartopy: 0.20.3
seaborn: None
numbagg: None
fsspec: 2022.5.0
cupy: None
pint: 0.19.2
sparse: None
setuptools: 63.2.0
pip: 22.1.2
conda: None
pytest: None
IPython: 7.33.0
sphinx: 5.0.2

wradlib: 1.16.0
</details>

kmuehlbauer, 2022-10-24:
Thanks @JulianGiles for raising this.

As you have already identified the problem, how about adding a PR? I think we could use the key names to avoid the hardcoded numbers.

JulianGiles, 2022-10-24:
I just created a PR with your suggestion, now the key names are used instead of indexes. I tried it with two stations, with and without the 'Station identifier' parameter and it is working. #597

kmuehlbauer, 2022-10-24:
fixed by #597

closed 2022-10-24 by kmuehlbauer