wradlib/wradlib #601: Question to radolan grid

goettlich, 2022-11-08:
Hello,

this is not a bug report but more of a question, as I am new to geodata and wradlib. Using the wrl.georefget_radolan_coordinates function, I get a slightly different result from what is mentioned in https://docs.wradlib.org/en/latest/generated/wradlib.georef.rect.get_radolan_grid.html:

```
import wradlib as wrl
proj_stereo = wrl.georef.create_osr("dwd-radolan")
radolan_coordinates = wrl.georef.get_radolan_coordinates(proj=proj_stereo, nrows=1100, ncols=900, wgs84=False)
print(f"\nLower Left : {radolan_coordinates[0][0]:.3f}, {radolan_coordinates[1][0]:.3f}")
print(f"Upper Right: {radolan_coordinates[0][899]:.3f}, {radolan_coordinates[1][1099]:.3f}")
```

The output that I get is the following:
Lower Left : -443.462, -4758.645
Upper Right: 455.538, -3659.645

The coordinates mentioned in the docs (https://docs.wradlib.org/en/latest/generated/wradlib.georef.rect.get_radolan_grid.html):
Lower Left : -443.4622, -4758.645
Upper Right: 456.5378, -3658.645

There is a small difference in both values of the upper right corner. Am I understanding something wrong here or is it just a typo in the docs?
The Wradlib version I am using is 1.16.2.

Thanks alot!

kmuehlbauer, 2022-11-08:
@goettlich Hi there, that might be a precision issue. What does your code show, if you change the formatting to `:.4f`?

kmuehlbauer, 2022-11-08:
Never mind, I got it now. I can think of that the one off mismatch comes from that in the one version the lower left corner of the upper right pixel is referenced and in the other the upper right corner of the upper right pixel

kmuehlbauer, 2022-11-08:
So, to rule out any misunderstanding.

The data from the docs reference what it states, the coordinates of the outer four corners (as is also shown like that in the DWD format description).

The grid coordinates you get back from the function reference the lower left points of each grid box.

That explains the difference of 1km for the upper right corner (1km grid spacing).

If you have a suggestion on how to make this more clear in the docs, just let us know.

kmuehlbauer, 2022-11-08:
Just for reference, here is a gist, where I played a bit with RADOLAN.

https://gist.github.com/kmuehlbauer/e050772fd0d5b7f88c71fff6a326f858

goettlich, 2022-11-09:
Aah this makes perfect sense! I think I overlooked the part in the brackets stating "lower left corner of every pixel", it is actually very clear. Thanks for the quick help :)

closed 2022-11-09 by goettlich