wradlib/wradlib #602: Add specified percentage parameter to clutter.histo_cut

overeem11, 2022-11-08:
This identification algorithm uses the histogram of temporal accumulated rainfall. It iteratively detects classes whose frequency falls below a specified percentage (1% by default) of the frequency of the class with the biggest frequency and remove the values from the dataset until the changes from iteration to iteration falls below a threshold. This specified percentage is currently hard-coded, which limits application of the algorithm.

#### MCVE Code Sample

```python
for i in range(index, -1, -1):
    if n[i] < (n[index] * 0.01):
          break

 for j in range(index, len(n)):
     if n[j] < (n[index] * 0.05):
           break
```

#### Expected Output
This cannot be reproduced with a few lines of code, but the output will remain the same if 0.01 would be supplied as a parameter (specified percentage) to the histo_cut function. Output will of course change when another value would be chosen.

#### Problem Description

Recommendation is to change the hard-coded default value of 0.01 (1%) in the histo_cut function in a parameter, so to add a parameter to wradlib.clutter.histo_cut. For OPERA radar data a value of 0.05 seems to work quite well, and this is currently only possible by changing the hard-coded value of 0.01 in 0.05 in the Python script "clutter.py" and by loading this script instead of loading the wradlib package. The 0.01 in the example above should then be replaced by a parameter, e.g. threshold, which should then be supplied as argument to the histo_cut function. The parameter would then occur in the two for loopes provided above.

#### Version

<details><summary>Output of <tt>wrl.show_versions()</tt></summary>

</details>

kmuehlbauer, 2022-11-09:
Thanks for opening this feature request @overeem11. That seems totally reasonable to make that value available as parameter.

One question though, would it make sense to provide two parameter, one for each loop (upper/lower bound)?

I think the code could also need some refreshing. I hope to free some cycles to get this done soon.

overeem11, 2022-11-09:
Thanks. Good idea to make a two parameter solution.

kmuehlbauer, 2022-11-09:
@overeem11 I've had a first take on that and I'm asking myself, if it would be of interest to be able to separate shading and clutter.

Currently we get back a combined mask, but we could think of returning separate masks for shading and clutter. Or we could return a mask with values like 

- 0 - rain
- 1 - clutter
- 2 - shading

Would that be of interest? If so we might need to think about backwards compatibility.

kmuehlbauer, 2022-11-09:
@overeem11 I'll add the parameterizations to start with, see #603. We can iterate on the above ideas/suggestions any time.

overeem11, 2022-11-09:
I think that would be interesting. In terms of backwards compatibility: the default setting could be to only return the combined mask for shading and clutter. Another parameter could be added to the histo_cut function, e.g. separate_mask. When this is supplied a mask for shading and one for clutter would be provided as output.

closed 2022-11-09 by kmuehlbauer

kmuehlbauer, 2022-11-09:
@overeem11 I'm planning to get wradlib 1.18 released with these changes later this week. Have a look at the documentation on histo_cut:

- https://docs.wradlib.org/en/latest/notebooks/classify/wradlib_histo_cut_example.html
- https://docs.wradlib.org/en/latest/generated/wradlib.clutter.histo_cut.html

overeem11, 2022-11-10:
Thanks. Regarding https://docs.wradlib.org/en/latest/notebooks/classify/wradlib_histo_cut_example.html: I notice that mask.astype(np.bool)  is used when plotting the mask, which works fine to show the clutter and shading areas. Are you sure that mask directly works when it it used to select clutter & shadings in data? Or should users also add .astype(np.bool) themselves when they start working with the radar data? Minor detail: "acummulation" > "accumulation".

Regarding https://docs.wradlib.org/en/latest/generated/wradlib.clutter.histo_cut.html some minor details:

- "and shadings set 2" > "and shadings set to 2".

- Perhaps mention the backwards compatability (or if not, that users should add .astype(np.bool))

  *   https://docs.wradlib.org/en/latest/notebooks/classify/wradlib_histo_cut_example.html
  *   https://docs.wradlib.org/en/latest/generated/wradlib.clutter.histo_cut.html

kmuehlbauer, 2022-11-10:
@overeem11 Thanks for taking the time.

You are right, if users want to have the same behavior in plotting, they would need to cast to bool. I'll add a note in the docs.

For workflows which rely on something like:

```
rain = np.where(~mask)
clutter = np.where(mask)
```
that should work without explicit casting.

overeem11, 2022-11-10:
Thanks for explaining.

PR #603 (2022-11-09): Histo cut and xarray backends
This PR adds parametrization to wradlib.clutter.histo_cut and adds description to wradlib xarray backends.

Fixes: #602