wradlib/wradlib #607: Unable to open RADOLAN files older than 2015, 1, 1

meteoDaniel, 2022-11-25:
I am using downloader from [@wetterdienst](https://github.com/earthobservations/wetterdienst) to receive radolan files. 

Unfortunaetyl I am not able to open files older than 1.1.2015 

```python
    start_date = datetime(2014, 1, 1)
    end_date = datetime(2014, 2, 1)
    radolan = DwdRadarValues(
        parameter=DwdRadarParameter.RADOLAN_CDC,
        start_date=start_date + timedelta(hours=1),
        end_date=end_date,
        resolution=DwdRadarResolution.HOURLY,
    )
    downloaded_data = []
    downloaded_attr = []
    for item in radolan.query():
        _data, _attr = wrl.io.read_radolan_composite(item.data)
        downloaded_data.append(_data)
        downloaded_attr.append(_attr)

```
I am receiving `UnicodeDecodeError`: 

```python
UnicodeDecodeError                        Traceback (most recent call last)
/app/src/radar_data/german_radar_data_download.py in <module>
    150         time_dimensions=["time"],
    151     )
--> 152 
    153 process_radolan_data_upload_historic(
    154     datetime(2014,1,1),

/app/src/radar_data/german_radar_data_download.py in process_radolan_data_upload_historic(dt_start, dt_end)
    107     dt_start = dt_start.replace(day=1)
    108     while dt_start < dt_end:
--> 109         print(dt_start)
    110         ds = process_radolan_data_download(
    111             dt_start,

/app/src/radar_data/german_radar_data_download.py in process_radolan_data_download(start_date, end_date, resolution, parameter)
     74         from IPython import embed
     75         embed()
---> 76         wrl.io.radolan.open_radolan_dataset(item.data)
     77         _data, _attr = wrl.io.read_radolan_composite(item.data)
     78         downloaded_data.append(_data)

/usr/local/lib/python3.9/site-packages/wradlib/io/radolan.py in read_radolan_composite(f, missing, loaddata, fillmissing)
    792 
    793     # get _radolan_file class
--> 794     with _radolan_file(
    795         f, fillmissing=fillmissing, copy=True, ancillary=False
    796     ) as radfile:

/usr/local/lib/python3.9/site-packages/wradlib/io/radolan.py in __init__(self, filename, fillmissing, copy, ancillary)
   1058         self._fill = fillmissing
   1059         self._copy = copy
-> 1060         if self.attrs["producttype"] == "ascii":
   1061             self._product = self.filename[0:2].upper()
   1062         self._ancillary = self._get_ancillary(requested=ancillary)

/usr/local/lib/python3.9/site-packages/wradlib/io/radolan.py in attrs(self)
   1067             # move file pointer to start
   1068             self.fp.seek(0)
-> 1069             self._attrs = self._read_attrs()
   1070         return self._attrs
   1071 

/usr/local/lib/python3.9/site-packages/wradlib/io/radolan.py in _read_attrs(self)
   1106 
   1107     def _read_attrs(self):
-> 1108         header = read_radolan_header(self.fp)
   1109         if isinstance(header, dict) and header["producttype"] == "ascii":
   1110             header["ncol"] = header.pop("ncols")

/usr/local/lib/python3.9/site-packages/wradlib/io/radolan.py in read_radolan_header(fid)
    716         if mychar == b"\x03":
    717             break
--> 718         header += str(mychar.decode())
    719     return header
    720 

UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8b in position 0: invalid start byte
``` 

wradlib==1.18.0
wetterdienst==0.48.0

kmuehlbauer, 2022-11-25:
I cannot reproduce with wradlib 1.18 and either bin or ascii data files downloaded from:

https://opendata.dwd.de/climate_environment/CDC/grids_germany/hourly/radolan/historical/bin/2014/

https://opendata.dwd.de/climate_environment/CDC/grids_germany/hourly/radolan/historical/ascii/2014/

```python
import xarray as xr

fname = "/home/kai/daten/daten/DWD/RW201401/raa01-rw_10000-1401010050-dwd---bin.gz"
ds = xr.open_dataset(fname, engine="radolan")
print(ds)
fname = "/home/kai/daten/daten/DWD/RW-20140101/RW_20140101-0050.asc"
ds = xr.open_dataset(fname, engine="radolan")
print(ds)
```

```
<xarray.Dataset>
Dimensions:  (y: 900, x: 900, time: 1)
Coordinates:
  * time     (time) datetime64[ns] 2014-01-01T00:50:00
  * y        (y) float64 -4.658e+06 -4.657e+06 ... -3.76e+06 -3.759e+06
  * x        (x) float64 -5.23e+05 -5.22e+05 -5.21e+05 ... 3.75e+05 3.76e+05
Data variables:
    RW       (y, x) float32 ...
Attributes:
    radarid:         10000
    formatversion:   3
    radolanversion:  2.13.1
    radarlocations:  ['asw', 'ham', 'emd', 'han', 'umd', 'ess', 'fld', 'drs',...
<xarray.Dataset>
Dimensions:  (y: 900, x: 900, time: 1)
Coordinates:
  * time     (time) datetime64[ns] 2014-01-01T00:50:00
  * y        (y) float64 -4.658e+06 -4.657e+06 ... -3.76e+06 -3.759e+06
  * x        (x) float64 -5.23e+05 -5.22e+05 -5.21e+05 ... 3.75e+05 3.76e+05
Data variables:
    /H       (y, x) float32 ...
Attributes:
    radarid:  10000
```

I'd report this at the wetterdienst issue-tracker. My gut feeling is, that something is going wrong when extracting and putting in the stream.

meteoDaniel, 2022-11-25:
Thanks @kmuehlbauer  for this support and fast response.

closed 2022-11-25 by meteoDaniel

kmuehlbauer, 2022-11-25:
@meteoDaniel You're welcome. Please ping me, if I can be of any help tracking down the error source.