wradlib/wradlib #615: DB_VELC data from IRIS/Sigmet files is read as int16 instead uint8

pygmypirate, 2023-02-02:
wradlib: 1.18.0
Python: 3.10.6
OS: Ubuntu 22.04

DB_VELC moment data (1-byte unfolded velocity) from IRIS/Sigmet raw data files is read by wradlib.io.iris.read_iris as **int16** instead **uint8**.  Currently, no decode function is used at all for DB_VELC, see [wradlib/io/iris.py](https://github.com/wradlib/wradlib/blob/87c1ee17a28bf48f9050f176f85ddf44955c04de/wradlib/io/iris.py#L1895).

Excerpt from IRIS Programming Guide[^1]:

> **4.4.42 1-byte Unfolded Velocity Format (DB_VELC)**
>
> This is the mean radial velocity corrected for both Nyquist folding and fall speed. It is scaled
> to cover a fixed span of +/- 75 meters/second. In data product files, the value 255 is used to
> indicate area not scanned.
> 
> 0 | Velocity data not available at this range
> 1 | -75.0 m/s (towards the radar)
> 2 | -74.4 m/s
> 128 | Zero velocity
> 129 | +0.6 m/s
> 254 | +75.0 m/s (away from the radar)
> 255 | Area not scanned

[^1]: Download: ftp://ftp.sigmet.com/outgoing/manuals/IRIS-Programming-Guide-M211318EN.pdf

kmuehlbauer, 2023-02-02:
Thanks @pygmypirate, this will be fixed over at [xradar](https://github.com/openradar/xradar). I'm currently in the process to refactor wradlib codebase to use xradar, so this might take some time.

See also https://github.com/wradlib/wradlib/issues/598

kmuehlbauer, 2023-02-09:
@pygmypirate That's now available in xradar 0.0.13.

closed 2023-02-09 by kmuehlbauer